The-OpenROAD-Project/OpenROAD #4196: [WARNING GRT-0350] Missing route to pin mem_issue_unit/io_dis_uops_1_bits_rob_idx[3].

oharboe, 2023-11-01:
### Describe the bug

unzip

run:

```
./run-me-BoomCore-asap7-base.sh
```

```
[deleted]
[WARNING GRT-0350] Missing route to pin mem_issue_unit/io_dis_uops_1_bits_rob_idx[3].
[WARNING GRT-0350] Missing route to pin mem_issue_unit/io_dis_uops_0_bits_rob_idx[3].
[WARNING GRT-0350] Missing route to pin mem_issue_unit/io_dis_uops_2_bits_rob_idx[3].
[WARNING GRT-0350] Missing route to pin mem_issue_unit/io_dis_uops_3_bits_rob_idx[3].
[WARNING GRT-0350] Missing route to pin rob/io_enq_uops_3_rob_idx[3].
[WARNING GRT-0350] Missing route to pin rob/io_enq_uops_0_rob_idx[3].
[WARNING GRT-0350] Missing route to pin rob/io_enq_uops_1_rob_idx[3].
[WARNING GRT-0350] Missing route to pin rob/io_enq_uops_2_rob_idx[3].
[WARNING GRT-0350] Missing route to pin mem_issue_unit/io_dis_uops_0_bits_rob_idx[2].
[WARNING GRT-0350] Missing route to pin mem_issue_unit/io_dis_uops_1_bits_rob_idx[2].
[WARNING GRT-0350] Missing route to pin mem_issue_unit/io_dis_uops_2_bits_rob_idx[2].
[WARNING GRT-0350] Missing route to pin mem_issue_unit/io_dis_uops_3_bits_rob_idx[2].
[deleted]
```

### Expected Behavior

@maliberty Said that GRT-0350 should never be observed.

### Environment

```shell
OpenROAD v2.0-10770-g6719c92e6
```

### To Reproduce

See above

eder-matheus, 2023-11-14:
@oharboe Could you try this branch: https://github.com/The-OpenROAD-Project/OpenROAD/pull/4263?

oharboe, 2023-11-14:
> @oharboe Could you try this branch: #4263?

I dont have this setup anymore, sorry. All I have left is this reproduction case in this issue.

eder-matheus, 2023-11-14:
> I dont have this setup anymore, sorry. All I have left is this reproduction case in this issue.

Np, we'll try rerunning the reproducible. It is just too large, so it sometimes freezes my local machine, but I'll try it in a better computer.

closed 2023-11-15 by eder-matheus

eder-matheus, 2023-11-17:
@oharboe Just a heads up: I've confirmed that the fix for this issue works. I don't see the warning messages in the GRT log anymore when running with the master branch.