The-OpenROAD-Project/OpenROAD #4197: Create a pass that reads in all macros and writes out io pin constraint files for all macros at top level

oharboe, 2023-11-01:
### Description

@louiic I have MegaBoom set up with asap7, but it fails in global routing of the top level.

The problem appears to be mainly that the pins are on the wrong sides of the macro.

[image]

### Suggested Solution

Add a command that can be run at the top level floorplan to create io pin placement constraint files for all the macros.

louiic, 2023-11-01:
Will take a look

oharboe, 2023-11-01:
Other than IO pin constraints, I was looking at the routing congestion.

From the global routing routing congestion heatmap and drilling down, I can see that the buffer tree that is having a lot of congestion is the `RegisterRead_1 iregister_read`

[image]

oharboe, 2023-11-02:
Here I can see which areas fail in global routing:

1. hold cell violations
2. around the ALU units that are not pulled out as macros.

[image]

[image]

closed 2023-11-02 by oharboe