The-OpenROAD-Project/OpenROAD #4209: PPL-0048 is too verbose, move from INFO to DEBUG

oharboe, 2023-11-03:
### Description

PPL-0048 is too verbose.....

```
[INFO PPL-0048] Restrict pins [{io_brinfo_cfi_type[0]} {io_brinfo_cfi_type[1]} {io_brinfo_cfi_type[2]} io_brinfo_mispredict {io_brinfo_pc_sel[0]} {io_brinfo_pc_sel[1]} io_brinfo_taken {io_brinfo_target_offset[0]} {io_brinfo_target_offset[1]} {io_brinfo_target_offset[2]} {io_brinfo_target_offset[3]} {io_brinfo_target_offset[4]} {io_brinfo_target_offset[5]} {io_brinfo_target_offset[6]} {io_brinfo_target_offset[7]} {io_brinfo_target_offset[8]} {io_brinfo_target_offset[9]} {io_brinfo_target_offset[10]} {io_brinfo_target_offset[11]} {io_brinfo_target_offset[12]} {io_brinfo_target_offset[13]} {io_brinfo_target_offset[14]} {io_brinfo_target_offset[15]} {io_brinfo_target_offset[16]} {io_brinfo_target_offset[17]} {io_brinfo_target_offset[18]} {io_brinfo_target_offset[19]} {io_brinfo_target_offset[20]} {io_brinfo_uop_br_mask[0]} {io_brinfo_uop_br_mask[1]} {io_brinfo_uop_br_mask[2]} {io_brinfo_uop_br_mask[3]} {io_brinfo_uop_br_mask[4]} {io_brinfo_uop_br_mask[5]} {io_brinfo_uop_br_mask[6]} {io_brinfo_uop_br_mask[7]} {io_brinfo_uop_br_mask[8]} {io_brinfo_uop_br_mask[9]} {io_brinfo_uop_br_mask[10]} {io_brinfo_uop_br_mask[11]} {io_brinfo_uop_br_mask[12]} {io_brinfo_uop_br_mask[13]} {io_brinfo_uop_br_mask[14]} {io_brinfo_uop_br_mask[15]} {io_brinfo_uop_br_mask[16]} {io_brinfo_uop_br_mask[17]} {io_brinfo_uop_br_mask[18]} {io_brinfo_uop_br_mask[19]} {io_brinfo_uop_br_tag[0]} {io_brinfo_uop_br_tag[1]} {io_brinfo_uop_br_tag[2]} {io_brinfo_uop_br_tag[3]} {io_brinfo_uop_br_tag[4]} io_brinfo_uop_edge_inst {io_brinfo_uop_ftq_idx[0]} {io_brinfo_uop_ftq_idx[1]} {io_brinfo_uop_ftq_idx[2]} {io_brinfo_uop_ftq_idx[3]} {io_brinfo_uop_ftq_idx[4]} {io_brinfo_uop_ftq_idx[5]} io_brinfo_uop_is_rvc {io_brinfo_uop_ldq_idx[0]} {io_brinfo_uop_ldq_idx[1]} {io_brinfo_uop_ldq_idx[2]} {io_brinfo_uop_ldq_idx[3]} {io_brinfo_uop_ldq_idx[4]} {io_brinfo_uop_pc_lob[0]} {io_brinfo_uop_pc_lob[1]} {io_brinfo_uop_pc_lob[2]} {io_brinfo_uop_pc_lob[3]} {io_brinfo_uop_pc_lob[4]} {io_brinfo_uop_pc_lob[5]} {io_brinfo_uop_rob_idx[0]} {io_brinfo_uop_rob_idx[1]} {io_brinfo_uop_rob_idx[2]} {io_brinfo_uop_rob_idx[3]} {io_brinfo_uop_rob_idx[4]} {io_brinfo_uop_rob_idx[5]} {io_brinfo_uop_rob_idx[6]} {io_brinfo_uop_stq_idx[0]} {io_brinfo_uop_stq_idx[1]} {io_brinfo_uop_stq_idx[2]} {io_brinfo_uop_stq_idx[3]} {io_brinfo_uop_stq_idx[4]} io_brinfo_valid {io_brupdate_b1_mispredict_mask[0]} {io_brupdate_b1_mispredict_mask[1]} {io_brupdate_b1_mispredict_mask[2]} {io_brupdate_b1_mispredict_mask[3]} {io_brupdate_b1_mispredict_mask[4]} {io_brupdate_b1_mispredict_mask[5]} {io_brupdate_b1_mispredict_mask[6]} {io_brupdate_b1_mispredict_mask[7]} {io_brupdate_b1_mispredict_mask[8]} {io_brupdate_b1_mispredict_mask[9]} {io_brupdate_b1_mispredict_mask[10]} {io_brupdate_b1_mispredict_mask[11]} {io_brupdate_b1_mispredict_mask[12]} {io_brupdate_b1_mispredict_mask[13]} {io_brupdate_b1_mispredict_mask[14]} {io_brupdate_b1_mispredict_mask[15]} {io_brupdate_b1_mispredict_mask[16]} {io_brupdate_b1_mispredict_mask[17]} {io_brupdate_b1_mispredict_mask[18]} {io_brupdate_b1_mispredict_mask[19]} {io_brupdate_b1_resolve_mask[0]} {io_brupdate_b1_resolve_mask[1]} {io_brupdate_b1_resolve_mask[2]} {io_brupdate_b1_resolve_mask[3]} {io_brupdate_b1_resolve_mask[4]} {io_brupdate_b1_resolve_mask[5]} {io_brupdate_b1_resolve_mask[6]} {io_brupdate_b1_resolve_mask[7]} {io_brupdate_b1_resolve_mask[8]} {io_brupdate_b1_resolve_mask[9]} {io_brupdate_b1_resolve_mask[10]} {io_brupdate_b1_resolve_mask[11]} {io_brupdate_b1_resolve_mask[12]} {io_brupdate_b1_resolve_mask[13]} {io_brupdate_b1_resolve_mask[14]} {io_brupdate_b1_resolve_mask[15]} {io_brupdate_b1_resolve_mask[16]} {io_brupdate_b1_resolve_mask[17]} {io_brupdate_b1_resolve_mask[18]} {io_brupdate_b1_resolve_mask[19]} {io_bypass_0_bits_data[0]} {io_bypass_0_bits_data[1]} {io_bypass_0_bits_data[2]} {io_bypass_0_bits_data[3]} {io_bypass_0_bits_data[4]} {io_bypass_0_bits_data[5]} {io_bypass_0_bits_data[6]} {io_bypass_0_bits_data[7]} {io_bypass_0_bits_data[8]} {io_bypass_0_bits_data[9]} {io_bypass_0_bits_data[10]} {io_bypass_0_bits_data[11]} {io_bypass_0_bits_data[12]} {io_bypass_0_bits_data[13]} {io_bypass_0_bits_data[14]} {io_bypass_0_bits_data[15]} {io_bypass_0_bits_data[16]} {io_bypass...
```

### Suggested Solution

This is too verbose for INFO level, move to DEBUG

rovinski, 2023-11-03:
Alternatively, maybe don't expand buses or print a count of the pins instead.

oharboe, 2023-11-03:
> Alternatively, maybe don't expand buses or print a count of the pins instead.

Both? Terse for INFO and verbose for DEBUG?

closed 2023-11-05 by maliberty