The-OpenROAD-Project/OpenROAD #4211: Metal spacing DRC error

oharboe, 2023-11-03:
### Describe the bug

I'm not sure what OpenROAD is trying to tell me here:

[image]

### Expected Behavior

No DRC error

### Environment

```shell
OpenROAD v2.0-10770-g6719c92e6
```

### To Reproduce

Coming up...

maliberty, 2023-11-03:
Is that the whole maker?  Are all shapes/nets visible?

oharboe, 2023-11-03:
> Is that the whole maker? Are all shapes/nets visible?

On second try, the snapshot looks like it has everything. I replaced the image and uploaded the `make detail_route_issue` file.

oharboe, 2023-11-03:
@maliberty Please confirm that this issue has what is needed to reproduce. Thanks!

maliberty, 2023-11-04:
You can plainly see the issue if you turn on only m4
[image]

maliberty, 2023-11-04:
There appears to be a usable m5 track between the macro and the power strap on m5.  However getting off that track is impossible due to via leading to an error

[image]

oharboe, 2023-11-04:
@osamahammad21 You should have access to https://drive.google.com/file/d/10ECM456C94DEikW51WINpRCTr4MvYN14/view?usp=sharing now

osamahammad21, 2023-11-06:
I did some investigation and I can say that it's a GRT problem. It's impossible for the router to avoid this violation:
Here is the route guides for the specified net:
[image]
At first glance I though it's a track assignment issue by assigning a bad track. But I find that the assigned track is the only valid track. Here is a screenshot for the available 3 tracks for that guide on Metal5
[image]
The net is stuck between the macro obstruction and the power grid on Metal5 and it turns out the middle track is the only one that doesn't cause a violation on Metal5.  The track next to the power strap is violating the SpacingTable rule which specifies the spacing to be 0.072 (because the strap is wide). The current spacing of the middle track is already 0.078 so using to the track to the left is not an option.
The router keeps moving the via up and down but it cannot escape the violation.

@maliberty I think @eder-matheus  is working on a via aware global routing which should cover this case https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1564