The-OpenROAD-Project/OpenROAD #4213: Problem in building OpenROAD-flow-scrips

rudro101, 2023-11-04:
### Describe the bug

I am using WSL2 method given in the official tutorial in website to build OpenROAD. I have followed every step and they were successful. But in the build step, the error is happening. I am using Ubuntu 22.04

### Expected Behavior

Successful build of openROAD

### Environment

```shell
[INFO FLW-0002] Updating git submodules.
[INFO FLW-0000] Using docker build method.[INFO FLW-0020] Building docker image for OpenROAD Flow.
Create docker image openroad/flow-centos7-dev:latest using docker/Dockerfile.dev
#0 building with "default" instance using docker driver

#1 [internal] load .dockerignore
#1 transferring context: 2B done
#1 DONE 0.0s

#2 [internal] load build definition from Dockerfile.dev
#2 transferring dockerfile: 505B done
#2 DONE 0.0s

#3 [internal] load metadata for docker.io/library/centos:centos7
#3 ...

#4 [auth] library/centos:pull token for registry-1.docker.io
#4 DONE 0.0s

#3 [internal] load metadata for docker.io/library/centos:centos7
#3 DONE 3.6s

#5 [1/5] FROM docker.io/library/centos:centos7@sha256:be65f488b7764ad3638f236b7b515b3678369a5124c47b8d32916d6487418ea4
#5 DONE 0.0s

#6 [internal] load build context
#6 transferring context: 23.38kB done
#6 DONE 0.0s

#7 [2/5] WORKDIR /tmp/installer/etc
#7 CACHED

#8 [3/5] COPY DependencyInstaller.sh .
#8 CACHED

#9 [4/5] COPY InstallerOpenROAD.sh        /tmp/installer/tools/OpenROAD/etc/DependencyInstaller.sh
#9 CACHED

#10 [5/5] RUN ./DependencyInstaller.sh       && rm -rf /tmp/installer
#10 CACHED

#11 exporting to image
#11 exporting layers done
#11 writing image sha256:fbd68ff596482cae2a92cf737115b8a15449c9fb306024805964ab3c23a44a9d done
#11 naming to docker.io/openroad/flow-centos7-dev:latest done
#11 DONE 0.0s

What's Next?
  View a summary of image vulnerabilities and recommendations → docker scout quickview
Create docker image openroad/flow-centos7-builder:latest using docker/Dockerfile.builder
#0 building with "default" instance using docker driver

#1 [internal] load build definition from Dockerfile.builder
#1 transferring dockerfile: 512B done
#1 DONE 0.0s

#2 [internal] load .dockerignore
#2 transferring context: 403B done
#2 DONE 0.0s
[... 146 lines cut ...]
 > [4/4] RUN ./build_openroad.sh --no_init --local --threads 12:
82.90 [ 25%] Building passes/opt/opt_mem_widen.o
83.24 [ 25%] Building passes/opt/opt_muxtree.o
87.35 [ 26%] Building passes/opt/opt_reduce.o
87.38 [ 26%] Building passes/opt/opt_dff.o
88.42 [ 26%] Building passes/opt/opt_share.o
193.7 gcc: fatal error: Killed signal terminated program cc1plus
193.7 compilation terminated.
193.7 make: *** [Makefile:704: kernel/satgen.o] Error 1
193.7 make: *** Waiting for unfinished jobs....
215.1 make: Leaving directory '/OpenROAD-flow-scripts/tools/yosys'
------
Dockerfile.builder:14
--------------------
  12 |     WORKDIR /OpenROAD-flow-scripts
  13 |     
  14 | >>> RUN ./build_openroad.sh --no_init --local --threads ${numThreads}
  15 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c ./build_openroad.sh --no_init --local --threads ${numThreads}" did not complete successfully: exit code: 2
```

### To Reproduce

NA

maliberty, 2023-11-05:
What version of gcc/g++ are you using?  Have you run the dependency installer?

rudro101, 2023-11-05:
[image]
I am using Docker method for building. Is it necessary to run dependency installer seperately? Didn't see that in tutorial.

stefanottili, 2023-11-06:
you're using --threads 12 in docker ... watch how much memory is used, try with --threads 1

luarss, 2023-11-07:
Also, if you are facing compilation issues with docker, you may try the prebuilt binaries method as listed [here](https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildWithDocker.html). The idea is to start a Ubuntu 22 (or one of our supported OSes) image and install the necessary binaries onto the docker image itself.

vvbandeira, 2023-11-27:
@rudro101, any updates? Were you able to compile with the suggestions above?

rudro101, 2023-11-28:
I didn't try pre binaries method so can't say about that. Previous
suggestions didn't work

closed 2023-11-28 by rudro101