The-OpenROAD-Project/OpenROAD #4228: properly displace ideal clocks in the gui timing report

maliberty, 2023-11-07:
### Describe the bug

When you have ideal clocks you have no meaningful data inside the clock tree.  In the text report_checks you get:

```
Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                  0.00    0.00    0.00   clock clk' (fall edge)
                          0.00    0.00   clock network delay (ideal)
                  0.00    0.00    0.00 v _44089_/CLK (DFFLQNx2_ASAP7_75t_R)
    18   42.48  144.41   98.39   98.39 ^ _44089_/QN (DFFLQNx2_ASAP7_75t_R)
```

However in the GUI you see phantom details:
[image]

### Expected Behavior

The GUI should suppress the data not shown by the text report for ideal clock nets.

### Environment

```shell
head of OR
```

### To Reproduce

The example is taken from asap7/aes

maliberty, 2023-11-07:
@gadfort if you have time for this let me know otherwise @AcKoucher can get to it when he does.

gadfort, 2023-11-07:
@maliberty I probably would not be able to get around to this for a while.

closed 2024-01-23 by maliberty