The-OpenROAD-Project/OpenROAD #4229: Need help understanding this error: `order_wires failed: net net61, shorts to another term at wire point …`

donn, 2023-11-07:
### Describe the bug

I've been trying to update an Odb-based antenna insertion routine- for some reason, this error pops up when I attempt OR's own antenna repair flow afterwards:

```
[INFO GRT-0018] Total wirelength: 153216 um
[INFO GRT-0014] Routed nets: 2287
[INFO GRT-0006] Repairing antennas, iteration 1.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[ERROR ODB-0391] order_wires failed: net net61, shorts to another term at wire point (268295 160320)
Error: antenna_repair.tcl, 23 ODB-0391
```

This may not be necessarily a bug. But I need help understanding the error- I don't understand how it "shorts to another term" as a virtual net that has not been routed yet.

I've tried inspecting it in the GUI here- regardless of the concerning amount of diodes for what doesn't look like a very long net- not really sure what qualifies as a "short to another term" here.

[image]

### Expected Behavior

A clearer explanation of what is wrong with the design would be helpful- but I'd also just appreciate help.

### Environment

```shell
95df42f3e443fe945902ce11b3e585f3e686e89e

(Once again note I'm using a Nix build script and ./etc/Env.sh does not take into account the Cmake flags I use.)
---
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Darwin 23.1.0
os: macOS 14.1
cmake version 3.24.3
CMake Warning at src/CMakeLists.txt:241 (message):
  spdlog: SPDLOG_FMT_EXTERNAL=ON

CMake Error at /nix/store/zzm5qbivbq13ffcsns8vy66w305k7m89-boost-1.79.0-dev/lib/cmake/Boost-1.79.0/BoostConfig.cmake:141 (find_package):
  Found package configuration file:

    /nix/store/zzm5qbivbq13ffcsns8vy66w305k7m89-boost-1.79.0-dev/lib/cmake/boost_serialization-1.79.0/boost_serialization-config.cmake

  but it set boost_serialization_FOUND to FALSE so package
  "boost_serialization" is considered to be NOT FOUND.  Reason given by
  package:

  No suitable build variant has been found.

  The following variants have been tried and rejected:

  * libboost_serialization.dylib (shared, Boost_USE_STATIC_LIBS=ON)

Call Stack (most recent call first):
  /nix/store/zzm5qbivbq13ffcsns8vy66w305k7m89-boost-1.79.0-dev/lib/cmake/Boost-1.79.0/BoostConfig.cmake:262 (boost_find_component)
  /nix/store/i472r5jaycpvb9rdhsynlz5vhsxhf2j1-cmake-3.24.3/share/cmake-3.24/Modules/FindBoost.cmake:594 (find_package)
  src/drt/CMakeLists.txt:49 (find_package)

-- The CXX compiler identification is Clang 11.1.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/26h0gs729p7gsr4ma0ar1ig4d042xr2r-clang-wrapper-11.1.0/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-10970-g95df42f3e
-- System name: Darwin
-- Compiler: Clang 11.1.0
-- Build type: RELEASE
-- Install prefix: /var/empty/local
[... 103 lines cut ...]
-- Install prefix: /var/empty/local
-- Found FLEX: /nix/store/7qmvihb1m9g1n3ba01x9y36lnra4hc8x-flex-2.6.4/bin/flex (found version "2.6.4") 
-- TCL library: TCL_LIBRARY-NOTFOUND
-- TCL header: TCL_HEADER-NOTFOUND
-- SSTA: 0
-- STA executable: /Users/donn/efabless/openroad/src/sta/app/sta
-- Found Protobuf: /nix/store/7knq04i3snwywjy0l9iwd7acqaypnhyv-protobuf-3.21.8/lib/libprotobuf.dylib (found version "3.21.8") 
-- Found re2: /nix/store/3x4d1dyc2zglgr3lsrrlq4gp2j336l4j-re2-2022-06-01-dev/lib/cmake/re2/re2Config.cmake (found version "9.0.0") 
-- Found GLPK: /nix/store/ycgylvh56r4q6h925qcvfi70dpi68az4-glpk-5.0/lib/libglpk.dylib  
-- Found OpenMP_CXX: -fopenmp=libomp (found version "5.0") 
-- Found OpenMP: TRUE (found version "5.0")  
-- GPU is not enabled
-- TCL library: TCL_LIBRARY-NOTFOUND
-- TCL header: TCL_HEADER-NOTFOUND
-- Found Eigen3: /nix/store/kiaz24ggfni8ydnkacz9bax93334ymdc-eigen-3.4.0/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- GUI is enabled
-- Charts widget is not enabled
-- Configuring incomplete, errors occurred!
```

### To Reproduce

Unpack, then run `run.sh` from the `reproducible` folder:

[repro.tgz](https://github.com/The-OpenROAD-Project/OpenROAD/files/13289490/repro.tgz)

eder-matheus, 2023-11-08:
@donn As Matt said, order_wires is a pain for us, especially in repair_antennas. What happens is that when evaluating the global routing for antenna violations, we transform the route guides into actual routing wires. This is needed because the antenna checker only works with routing wires, not the guide format.

Also, the antenna checker requires that order_wires be called for these temporary wires. The cause of the error you see is that this temporary routing can sometimes overlap with instance pins, and order_wires doesn't handle that well.

I made some improvements in the repair_antennas code to avoid such overlaps between tmp routing and pins, but it looks like it still requires more modifications. If the test case you see this issue can be shared, could you upload it here for investigation?

Btw, we also have a plan to make the antenna checker understand the route guides format, but this requires a large change on how the checker works.

donn, 2023-11-08:
Thanks for the explanation @eder-matheus! Will most likely have to just work around that, perhaps by… not inserting that many diodes.

I already uploaded the test case under "To Reproduce": link here again for your convenience. https://github.com/The-OpenROAD-Project/OpenROAD/files/13289490/repro.tgz

eder-matheus, 2023-11-08:
> Thanks for the explanation @eder-matheus! Will most likely have to just work around that, perhaps by… not inserting that many diodes.
>
> I already uploaded the test case under "To Reproduce": link [...]

Thanks very much! Sorry, I didn't notice the attached test case before (is almost 11pm here :upside_down_face:).

I'll look if I can fix the error in your test case, so a workaround would not be necessary in your end.

eder-matheus, 2023-11-09:
@donn I've found that there are a bunch of diodes overlapping in the position pointed out by the error message. I'm investigating now why the overlaps are occurring, it seems like an error from the diode insertion process.

maliberty, 2023-11-09:
@eder-matheus has the dpl been run?

eder-matheus, 2023-11-09:
> @eder-matheus has the dpl been run?

DPL runs after the violations are checked. The overlapped diodes should not exist in the first place, so I'm looking at why we have them.

donn, 2023-11-10:
@eder-matheus …May be the custom script.

The overlapping part is all the explanation I need though. In that scenario, I simply forgot to legalize. 🙃 

Thank you!