The-OpenROAD-Project/OpenROAD #4237: Attempting to `read_db` invalid files causes a crash

donn, 2023-11-09:
### Describe the bug

Upon attempting to read an invalid ODB file, OpenROAD doesn't attempt to validate it, and instead crashes with a cryptic message of `Error: read_db.tcl, 1 ios_base::clear: unspecified iostream_category error`

Ultimately, doesn't really affect functionality much, but a better user-facing message may be helpful from a usability perspective.

### Expected Behavior

Something along the lines of:
 
`[ERROR ORD-XXXX] Odb file is invalid…`

### Environment

```shell
Git commit: 5d50b1fc71594f9c94db324a1abe8f7fca44fc28
kernel: Darwin 23.1.0
os: macOS 14.1
cmake version 3.25.3
-- The CXX compiler identification is Clang 14.0.6
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/52yh0ynd87ihxhnpwa4snfvpcld833d9-clang-wrapper-14.0.6/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-11001-g5d50b1fc7
-- System name: Darwin
-- Compiler: Clang 14.0.6
-- Build type: RELEASE
-- Install prefix: /var/empty/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is Clang 14.0.6
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/52yh0ynd87ihxhnpwa4snfvpcld833d9-clang-wrapper-14.0.6/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/bin/python3.9 (found version "3.9.6") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
[... 122 lines cut ...]
 3# abort in /usr/lib/system/libsystem_c.dylib
 4# __cxxabiv1::__aligned_malloc_with_fallback(unsigned long) in /nix/store/fgjs9r7ps7xdm127sparlp2vqycajarn-libcxxabi-14.0.6/lib/libc++abi.1.0.dylib
 5# demangling_terminate_handler() in /nix/store/fgjs9r7ps7xdm127sparlp2vqycajarn-libcxxabi-14.0.6/lib/libc++abi.1.0.dylib
 6# std::__terminate(void (*)()) in /nix/store/fgjs9r7ps7xdm127sparlp2vqycajarn-libcxxabi-14.0.6/lib/libc++abi.1.0.dylib
 7# __cxa_get_exception_ptr in /nix/store/fgjs9r7ps7xdm127sparlp2vqycajarn-libcxxabi-14.0.6/lib/libc++abi.1.0.dylib
 8# __cxxabiv1::exception_cleanup_func(_Unwind_Reason_Code, _Unwind_Exception*) in /nix/store/fgjs9r7ps7xdm127sparlp2vqycajarn-libcxxabi-14.0.6/lib/libc++abi.1.0.dylib
 9# std::__1::__throw_failure(char const*) in /nix/store/spcsjq3liwrmz45a38mpg36yxf552r97-libcxx-14.0.6/lib/libc++.1.0.dylib
10# std::__1::__throw_failure(char const*) in /nix/store/spcsjq3liwrmz45a38mpg36yxf552r97-libcxx-14.0.6/lib/libc++.1.0.dylib
11# std::__1::basic_istream<char, std::__1::char_traits<char> >::read(char*, long) in /nix/store/spcsjq3liwrmz45a38mpg36yxf552r97-libcxx-14.0.6/lib/libc++.1.0.dylib
12# odb::operator>>(odb::dbIStream&, odb::_dbDatabase&) in /Users/donn/efabless/openroad/build/src/openroad
13# odb::dbDatabase::read(std::__1::basic_ifstream<char, std::__1::char_traits<char> >&) in /Users/donn/efabless/openroad/build/src/openroad
14# read_db(odb::dbDatabase*, char const*) in /Users/donn/efabless/openroad/build/src/openroad
15# _wrap_read_db(_object*, _object*) in /Users/donn/efabless/openroad/build/src/openroad
16# cfunction_call in /nix/store/cy4nsgrfg215s27dmmgrgb82j19i4j0m-python3-3.10.13/lib/libpython3.10.dylib
17# _PyObject_MakeTpCall in /nix/store/cy4nsgrfg215s27dmmgrgb82j19i4j0m-python3-3.10.13/lib/libpython3.10.dylib
18# call_function in /nix/store/cy4nsgrfg215s27dmmgrgb82j19i4j0m-python3-3.10.13/lib/libpython3.10.dylib
19# _PyEval_EvalFrameDefault in /nix/store/cy4nsgrfg215s27dmmgrgb82j19i4j0m-python3-3.10.13/lib/libpython3.10.dylib
20# _PyEval_Vector in /nix/store/cy4nsgrfg215s27dmmgrgb82j19i4j0m-python3-3.10.13/lib/libpython3.10.dylib
21# PyEval_EvalCode in /nix/store/cy4nsgrfg215s27dmmgrgb82j19i4j0m-python3-3.10.13/lib/libpython3.10.dylib
22# _PyRun_SimpleFileObject in /nix/store/cy4nsgrfg215s27dmmgrgb82j19i4j0m-python3-3.10.13/lib/libpython3.10.dylib
23# _PyRun_AnyFileObject in /nix/store/cy4nsgrfg215s27dmmgrgb82j19i4j0m-python3-3.10.13/lib/libpython3.10.dylib
24# Py_RunMain in /nix/store/cy4nsgrfg215s27dmmgrgb82j19i4j0m-python3-3.10.13/lib/libpython3.10.dylib
25# main in /Users/donn/efabless/openroad/build/src/openroad
Abort trap: 6
```

maliberty, 2023-11-09:
I'd advise against using low level odb APIs.  The db you are creating has no logger and can't issue any messages.  I can put more information in the exception but you would have to catch it yourself.

maliberty, 2023-11-09:
This is preferrable 
```
from openroad import Tech, Design

tech = Tech()
design = Design(tech)
design.readDb("my_odb.odb")
```

PR #4238 (2023-11-09): odb/ord: improve error messages on invalid db load
Fixes #4237
commit 8642e170 (2023-11-09): odb/ord: improve error messages on invalid db load

Fixes #4237
closed 2023-11-09 by maliberty