The-OpenROAD-Project/OpenROAD #4242: mpl2 segfault

oharboe, 2023-11-09:
### Describe the bug

Crash in mpl2:

```
 0# 0x00005615FB05A6F3 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007F21E6E3C460 in /lib/x86_64-linux-gnu/libc.so.6
 2# mpl2::SoftMacro::getPinX() const in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 3# mpl2::HierRTLMP::setTemporaryStdCellLocation(mpl2::Cluster*, odb::dbInst*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 4# mpl2::HierRTLMP::setModuleStdCellsLocation(mpl2::Cluster*, odb::dbModule*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# mpl2::HierRTLMP::generateTemporaryStdCellsPlacement(mpl2::Cluster*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# mpl2::HierRTLMP::generateTemporaryStdCellsPlacement(mpl2::Cluster*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# mpl2::HierRTLMP::generateTemporaryStdCellsPlacement(mpl2::Cluster*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# mpl2::HierRTLMP::hierRTLMacroPlacer() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 9# mpl2::MacroPlacer2::place(int, int, int, int, int, float, int, float, int, int, int, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, int, bool, char const*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
10# mpl2::rtl_macro_placer_cmd(int, int, int, int, float, int, float, int, int, int, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, float, int, bool, char const*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
11# 0x00005615FCA082DD in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
12# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
13# 0x00007F21EB486C43 in /lib/x86_64-linux-gnu/libtcl8.6.so
14# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
15# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
16# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
17# ord::tclAppInit(Tcl_Interp*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
18# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
19# main in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
20# 0x00007F21E6E23A90 in /lib/x86_64-linux-gnu/libc.so.6
21# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
22# _start in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
```

### Expected Behavior

No crash

### Environment

```shell
coming up
```

### To Reproduce

coming up

QuantamHD, 2023-11-09:
Probably needs a test case of some kind @oharboe

oharboe, 2023-11-09:
> Probably needs a test case of some kind @oharboe

Yes, working on it. I just needed to save some stuff while I had it...

oharboe, 2023-11-09:
Hmm... Tricky to create testcase than I thought. I will reopen if I manage to do so...

oharboe, 2023-11-10:
Whittling down:

```
openroad -python ~/OpenROAD-flow-scripts/tools/OpenROAD/etc/deltaDebug.py --persistence 6 --use_stdout --error_string "grt::FastRouteCore::copyGrids" --base_db_path build/results/asap7/RegisterFileSynthesizable_1/base/4_cts.odb --step "./run-me-RegisterFileSynthesizable_1-asap7-base.sh"
```

oharboe, 2023-11-10:
@maliberty Please confirm that you can reproduce. Ca. 15 minutes on my PC to reproduce the crash.

oharboe, 2023-11-12:
@maliberty Please confirm that you can reproduce. Ca. 2 minutes on my PC to reproduce the crash.

oharboe, 2023-11-12:
[image]
[image]

oharboe, 2023-11-12:
@maliberty Please confirm that you can reproduce. Ca. 1 minutes on my PC to reproduce the crash. The google drive link(33mBytes, too big for github) is available to anyone with the link.

oharboe, 2023-11-12:
Classic placement does produce a floorplan without crashing.

[image]

closed 2023-11-13 by maliberty

AcKoucher, 2024-07-03:
@oharboe It looks like the link for the test case is broken

PR #4255 (2023-11-13): mpl2: fix crash in HierRTLMP::setTemporaryStdCellLocation
More of a quick fix than a root cause fix.

Fixes #4242
commit d897e91c (2023-11-13): mpl2: fix crash in HierRTLMP::setTemporaryStdCellLocation

More of a quick fix than a root cause fix.

Fixes #4242