The-OpenROAD-Project/OpenROAD #4246: [WARNING ODB-0177] error: undefined via (via5_6_120_288_1_2_58_322) referenced

oharboe, 2023-11-10:
### Describe the bug

I have created mock abstracts using only the floorplan.odb/sdc information so that I can iterate quickly on the MegaBoom floorplan. However, I get an error message in `floorplan.tcl` and warnings:

unzip [floorplan.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/13319173/floorplan.zip)

```
$ ./run-me-BoomTile-asap7-base.sh 
vars-BoomTile-asap7-base.sh: line 25: export: `--no-print-directory': not a valid identifier
OpenROAD v2.0-10996-g2bdae1c13 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /home/oyvind/megaboom/bar/platforms/asap7/lef/asap7_tech_1x_201209.lef
[INFO ODB-0223]     Created 24 technology layers
[INFO ODB-0224]     Created 9 technology vias
[INFO ODB-0226] Finished LEF file:  /home/oyvind/megaboom/bar/platforms/asap7/lef/asap7_tech_1x_201209.lef
[INFO ODB-0222] Reading LEF file: /home/oyvind/megaboom/bar/platforms/asap7/lef/asap7sc7p5t_28_R_1x_220121a.lef
[INFO ODB-0225]     Created 212 library cells
[INFO ODB-0226] Finished LEF file:  /home/oyvind/megaboom/bar/platforms/asap7/lef/asap7sc7p5t_28_R_1x_220121a.lef
[INFO ODB-0222] Reading LEF file: bazel-out/k8-fastbuild/bin/build/results/asap7/table_256x12/base/table_256x12.lef
[WARNING ODB-0177] error: undefined via (via5_6_120_288_1_2_58_322) referenced
[WARNING ODB-0177] error: undefined via (via5_6_120_288_1_2_58_322) referenced
[deleted]
```

### Expected Behavior

I should get a mock floorplan that I can iterate on.

### Environment

```shell
OpenROAD v2.0-10996-g2bdae1c13
```

### To Reproduce

See above

maliberty, 2023-11-10:
How did you generate the odb?

oharboe, 2023-11-10:
> How did you generate the odb?

Which .odb?

floorplan.tcl takes abstracts and 1_synth.v file as inputs, no?

maliberty, 2023-11-11:
I don't know where table_256x12.lef came from but it contains references to this undefined via which is correctly an error.

oharboe, 2023-11-11:
I will see if I can create a reproduction case for the illegal lef

closed 2023-11-11 by oharboe