The-OpenROAD-Project/OpenROAD #4252: support LEF58_MINWIDTH property "MINWIDTH 1.0 WRONGDIRECTION ;"

stefanottili, 2023-11-12:
### Describe the bug

test.or:
```
read_lef test.lef
read_def test.def -continue_on_errors
```
openroad -gui def2gds/wrongdirection.or
```
OpenROAD v2.0-11034-g2c985025e 
[INFO ODB-0222] Reading LEF file: test.lef
[INFO ODB-0388] unsupported LEF58_MINWIDTH property for layer M4 :"MINWIDTH 1.0 WRONGDIRECTION ; "
[INFO ODB-0223]     Created 3 technology layers
[INFO ODB-0224]     Created 1 technology vias
[INFO ODB-0226] Finished LEF file:  test.lef
[INFO ODB-0127] Reading DEF file: test.def
[INFO ODB-0128] Design: test
[INFO ODB-0133]     Created 1 nets and 0 connections.
[INFO ODB-0134] Finished DEF file: test.def
```

### Expected Behavior

support LEF58_MINWIDTH property "MINWIDTH 1.0 WRONGDIRECTION ;"

### Environment

```shell
OpenROAD v2.0-11034-g2c985025e
```

### To Reproduce

[wrongdirection.tgz](https://github.com/The-OpenROAD-Project/OpenROAD/files/13327055/wrongdirection.tgz)

maliberty, 2023-11-12:
Do you have an actual PDK that requires this or is it just a nice to have?

stefanottili, 2023-11-12:
This is a testcase from klayout, so it’s a real PDK, but none that would currently use OpenRoad.
But since this property seems to be a subset of WIDTHTABLE that OpenRoad does support, it should be easy to implement, no ? Just one element in the WRONGDIRECTION table.

maliberty, 2023-11-12:
Its a matter of priority - there are a lot of rules that we need to work on.

maliberty, 2023-12-20:
@stefanottili is this an old pdk?  I see the syntax in a 2012 manual but not in a later one which makes me think it was dropped at some point.

maliberty, 2023-12-20:
It looks like this got replaced with PROPERTY LEF58_WIDTH "WIDTH minWidth [WRONGDIRECTION] ; "

closed 2026-08-15 by maliberty

stefanottili, 2026-08-15:
It parses, but without having tried it: does it have an effect on the display or extraction ?
Should that be a separate issue or keep this issue open ?