The-OpenROAD-Project/OpenROAD #4256: mpl2 fails to place macros where classic macro placement succeeds

oharboe, 2023-11-13:
### Description

Classic succeeds: [image]

mpl2 fails.

```
MultiLevelMacroPlacement] sa_id = 188, target_util = 1.1500001,  target_dead_space = 0.90000004
[MultiLevelMacroPlacement] sa_id = 189, target_util = 1.1500001,  target_dead_space = 0.95000005
[INFO MPL-0032] [MultiLevelMacroPlacement] Working on children of cluster: ((root)_glue_logic_0)_glue_logic_0_0_0_0, Outline 2.052, 2.16  3536.136, 2356.588
[EnhancedMacroPlacement] Simulated Annealing Summary for cluster ((root)_glue_logic_0)_glue_logic_0_0_0_0
[EnhancedMacroPlacement] sa_id = 0, target_util = 1000000.0,  target_dead_space = 0.99999
[ERROR MPL-2067] [MultiLevelMacroPlacement] Failed on cluster ((root)_glue_logic_0)_glue_logic_0_0_0_0
Error: macro_place_util.tcl, 138 MPL-2067
Command exited with non-zero status 1
Elapsed time: 36:40.00[h:]min:sec. CPU time: user 15145.64 sys 389.73 (706%). Peak memory: 10083828KB.
make[1]: *** [/home/oyvind/OpenROAD-flow-scripts/flow/Makefile:593: do-2_4_floorplan_macro] Error 1
make: *** [/home/oyvind/OpenROAD-flow-scripts/flow/Makefile:613: do-floorplan] Error 2
```

To reproduce, unzip https://drive.google.com/file/d/17HgnvWMQBAORIPfR6UOaHqXaJ064rrMi/view?usp=sharing and run:

```
NUM_CORES=16 ./run-me-BoomTile-asap7-base.sh
```

### Suggested Solution

Should place macros

maliberty, 2023-11-13:
Does the fix for #4242 work here?

oharboe, 2023-11-14:
@maliberty Please confirm that you can repeat the problem. deltaDebug did not produce a faster failure, so the reproduction case is the above.

AcKoucher, 2024-07-03:
@oharboe It looks the file for the test case doesn't exist anymore in the link.

oharboe, 2024-07-03:
@AcKoucher I haven't seen this kind of crash in the last half a year, so no worries :-)

AcKoucher, 2024-07-03:
Close this one then?

closed 2024-07-03 by oharboe