The-OpenROAD-Project/OpenROAD #4267: Improve advice from -  [ERROR GPL-0302] Use a higher -density or re-floorplan with a larger core area.

oharboe, 2023-11-15:
### Description

Alternative, what is the actual and suggested CORE_UTILIZATION?

```
[ERROR GPL-0302] Use a higher -density or re-floorplan with a larger core area.
Given target density: 0.10
Suggested target density: 0.14
Error: global_place_skip_io.tcl, 30 GPL-0302
Command exited with non-zero status 1
Elapsed time: 0:02.44[h:]min:sec. CPU time: user 2.10 sys 0.34 (100%). Peak memory: 601048KB.
```

### Suggested Solution

List is the actual and suggested CORE_UTILIZATION.

agostini01, 2025-05-21:
Hi, I ran into this issue when trying a nangate45 design (selected utilization: 10%, suggested utilization: 11%). I originally thought that when aiming for a CORE_UTILIZATION target, the core area would resize to match target utilization. 

Isn't it the case? Should this be a silent warning?

rovinski, 2025-05-30:
The utilization matches what you set it to when the design is loaded, but the utilization changes when running optimizations and adding/removing/resizing cells.

gudeh, 2025-07-09:
Hi, this error 302 was modified, now it is a warning. This error happened when the provided density was too low. Now gpl automatically modified the density for a valid value. The change was performed in PR https://github.com/The-OpenROAD-Project/OpenROAD/pull/7692 

@oharboe do you still want to see the CORE_UTILIZATION value with the new warning, or perhaps somewhere else in gpl's log?

oharboe, 2025-07-09:
It is a usability issue that is not very high on my list: how do I convert from the units in the warning to CORE_UTILIZATION units?

Perhaps it would be better to fix this in ORFS by introducing a CORE_DENSITY variable that is an alternative to CORE_UTILIZATION.

Why do we have two units here anyway?

gudeh, 2025-07-09:
The PLACE_DENSITY is used in gpl as the target density. The target density is an upper limit for the gpl bins. In practice the value is used for scaling the bin areas. Setting this value does not guarantee to reach such density on bins, in fact it just an upper limit.

As for CORE_UTILIZATION, I understand it is a general or "global" setting that is not directly used by GPL. Instead, it is used by MPL, although I am not exactly sure how MPL utilizes it.

maliberty, 2026-03-24:
This issue has been automatically marked as stale because it has not had recent activity. It will be closed in 21 days if no further activity occurs. Remove the Stale label or comment to keep it open.