The-OpenROAD-Project/OpenROAD #4268: grt::GlobalRouter::getAllPorts() segfault

oharboe, 2023-11-15:
### Describe the bug

```
[INFO RSZ-0046] Found 4395 endpoints with hold violations.
Signal 11 received
Stack trace:
 0# 0x000055FBDDC73C73 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007F3C61C3C460 in /lib/x86_64-linux-gnu/libc.so.6
 2# grt::GlobalRouter::getAllPorts() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 3# grt::GlobalRouter::checkPinPlacement() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 4# grt::GlobalRouter::initNets(std::vector<grt::Net*, std::allocator<grt::Net*> >&) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# grt::GlobalRouter::updateDirtyRoutes(bool) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# rsz::Resizer::updateParasitics(bool) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# rsz::RepairHold::repairHoldPass(sta::Vector<sta::Vertex*>&, sta::LibertyCell*, double, double, bool, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# rsz::RepairHold::repairHold(sta::Vector<sta::Vertex*>&, sta::LibertyCell*, double, double, bool, int, int, bool) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 9# rsz::RepairHold::repairHold(double, double, bool, float, int, bool) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
10# 0x000055FBDDDDD4C8 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
11# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
12# 0x00007F3C6641FC43 in /lib/x86_64-linux-gnu/libtcl8.6.so
13# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
14# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
15# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
16# ord::tclAppInit(Tcl_Interp*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
17# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
18# main in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
19# 0x00007F3C61C23A90 in /lib/x86_64-linux-gnu/libc.so.6
20# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
21# _start in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad

```

### Expected Behavior

No crash

### Environment

```shell
OpenROAD v2.0-11072-g2d83b7137
```

### To Reproduce

Coming up.

oharboe, 2023-11-15:
Whittling down...

```
openroad -python ~/OpenROAD-flow-scripts/tools/OpenROAD/etc/deltaDebug.py --persistence 4 --use_stdout --error_string "[MPL-2067](grt::GlobalRouter::getAllPorts())" --base_db_path bazel-out/k8-fastbuild/bin//build/results/asap7/regfile_128x65/base/3_place.odb "./run-me-regfile_128x65-asap7-base.sh"
```

oharboe, 2023-11-15:
I have seen a different SEGFAULT stack trace, for the same test case, but I'm going to assume it is the same problem and retest after fix...

```
[INFO RSZ-0046] Found 4395 endpoints with hold violations.
Signal 11 received
Stack trace:
 0# 0x0000557DE698FC73 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007F40C343C460 in /lib/x86_64-linux-gnu/libc.so.6
 2# grt::GlobalRouter::getAllPorts() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 3# grt::GlobalRouter::checkPinPlacement() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 4# grt::GlobalRouter::initNets(std::vector<grt::Net*, std::allocator<grt::Net*> >&) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# grt::GlobalRouter::updateDirtyRoutes(bool) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# rsz::Resizer::updateParasitics(bool) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# rsz::RepairHold::repairHoldPass(sta::Vector<sta::Vertex*>&, sta::LibertyCell*, double, double, bool, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# rsz::RepairHold::repairHold(sta::Vector<sta::Vertex*>&, sta::LibertyCell*, double, double, bool, int, int, bool) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 9# rsz::RepairHold::repairHold(double, double, bool, float, int, bool) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
10# 0x0000557DE6AF94C8 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
11# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
12# 0x00007F40C7B6CC43 in /lib/x86_64-linux-gnu/libtcl8.6.so
13# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
14# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
15# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
16# ord::tclAppInit(Tcl_Interp*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
17# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
18# main in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
19# 0x00007F40C3423A90 in /lib/x86_64-linux-gnu/libc.so.6
20# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
21# _start in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
Command terminated by signal 11
```

oharboe, 2023-11-17:
A slightly faster test-case, perhaps 5 minutes?

[crash2.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/13387909/crash2.zip)

arthurjolo, 2023-11-17:
@oharboe I am not sure if is something with my computer but I can't extract the files from this faster test-case, the original I was able to extract, but the faster one I can't.

oharboe, 2023-11-17:
> @oharboe I am not sure if is something with my computer but I can't extract the files from this faster test-case, the original I was able to extract, but the faster one I can't.

I compressed with zip -Zbzip2, so depending on your OS, you may need to install bzip2 seperately from zip.

arthurjolo, 2023-11-17:
O

> > @oharboe I am not sure if is something with my computer but I can't extract the files from this faster test-case, the original I was able to extract, but the faster one I can't.
> 
> I compressed with zip -Zbzip2, so depending on your OS, you may need to install bzip2 seperately from zip.

I will try that, thanks.

oharboe, 2023-11-22:
@arthurjolo I think it might be worth fixing #4297 first, because it has a much shorter turnaround time, then see if this is fixed.

arthurjolo, 2023-11-22:
> @arthurjolo I think it might be worth fixing #4297 first, because it has a much shorter turnaround time, then see if this is fixed.

I have found what it causing this segfault, I am working on a solution. I will test for [#4297](https://github.com/The-OpenROAD-Project/OpenROAD/issues/4297) as well when I complete the solution.

closed 2023-12-01 by eder-matheus