The-OpenROAD-Project/OpenROAD #4277: Crash on clock tree synthesis step with combinatorial circuit

lustefan97, 2023-11-17:
### Describe the bug

I have a dummy adder circuit in which there is no clock, but I use the same script on all my designs, sequential of not. I had no issue before, but now OpenROAD crashes on the CTS step with that design.

Here is the log : 

```
Signal 11 received
Stack trace:
 0# 0x000055F99EB4BDBB in openroad
 1# 0x00007F2E5AA42520 in /lib/x86_64-linux-gnu/libc.so.6
 2# sta::LibertyCell::bufferPorts(sta::LibertyPort*&, sta::LibertyPort*&) const in openroad
 3# cts::TritonCTS::selectRootBuffer(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&) in openroad
 4# cts::TritonCTS::setRootBuffer(char const*) in openroad
 5# 0x000055F99F1B98AC in openroad
 6# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 7# 0x00007F2E5EF27028 in /lib/x86_64-linux-gnu/libtcl8.6.so
 8# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
 9# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
10# gui::TclCmdInputWidget::executeCommand(QString const&, bool, bool) in openroad
11# gui::CmdInputWidget::handleEnterKeyPress(QKeyEvent*) in openroad
12# gui::TclCmdInputWidget::keyPressEvent(QKeyEvent*) in openroad
13# QWidget::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
14# QFrame::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
15# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
16# QApplication::notify(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
17# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
18# 0x00007F2E5C7CC02F in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
19# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
20# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
21# QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
22# 0x00007F2E540130B4 in /usr/lib/x86_64-linux-gnu/qt5/plugins/platforminputcontexts/libibusplatforminputcontextplugin.so
23# 0x00007F2E5BAF1793 in /lib/x86_64-linux-gnu/libQt5Core.so.5
24# QDBusPendingCallWatcher::finished(QDBusPendingCallWatcher*) in /lib/x86_64-linux-gnu/libQt5DBus.so.5
25# QObject::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
26# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
27# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
28# QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
29# 0x00007F2E5BB13A67 in /lib/x86_64-linux-gnu/libQt5Core.so.5
30# g_main_context_dispatch in /lib/x86_64-linux-gnu/libglib-2.0.so.0
31# 0x00007F2E5AD71258 in /lib/x86_64-linux-gnu/libglib-2.0.so.0
32# g_main_context_iteration in /lib/x86_64-linux-gnu/libglib-2.0.so.0
33# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
34# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
35# QCoreApplication::exec() in /lib/x86_64-linux-gnu/libQt5Core.so.5
36# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) in openroad
37# ord::tclAppInit(Tcl_Interp*) in openroad
38# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
39# main in openroad
40# 0x00007F2E5AA29D90 in /lib/x86_64-linux-gnu/libc.so.6
41# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
42# _start in openroad
Segmentation fault (core dumped)
```

### Expected Behavior

No crashes, maybe just a warning as there is no clock tree to synthesize

### Environment

```shell
Git commit: 4759105234654bd6e30af9179e9523c38ede62a5
kernel: Linux 6.2.0-36-generic
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
```

### To Reproduce

[TestCase.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/13392305/TestCase.zip)

Extract this directory to _OpenROAD-flow-scripts/tools/OpenROAD/test/_, enter the _TestCase_ directory, use `openroad -gui` to enter the gui and use `source "./scripts/constant_density/10_adder_diedim_130_pd_0.7/adder_128_asap7.tcl"` in the TCL command box

precisionmoon, 2023-11-17:
Thanks for reporting the issue.  Could you help with some missing files so we can look into this quickly?

../helpers.tcl
../flow_helpers.tcl

precisionmoon, 2023-11-17:
I can avoid the errors by extracting the testcase under OpenROAD-flow-scripts/tools/OpenROAD/test.   Let me look into the issue.

lustefan97, 2023-11-23:
@precisionmoon Hi, any news ?

precisionmoon, 2023-11-27:
I fixed this issue as part of another PR 1888.   The PR is under review now and under validation with CI tests.

lustefan97, 2023-12-05:
Yes, perfect thank you !

closed 2023-12-05 by lustefan97

precisionmoon, 2023-12-12:
PR 4330 fixed this issue.