The-OpenROAD-Project/OpenROAD #4292: CTS step is taking unusually long time - 2550 secondns

sandeep-amd, 2023-11-21:
### Describe the bug

I have 2 similar system . On one system CTS is taking around 625 seconds while on the other system  it is taking 2550 seconds. These results are consistent.  

On both system cpufreq governor is set to performance.

**System 1 log: (625 seconds)**

Repair setup and hold violations...
TNS end percent 100
Skipping gate cloning during optimization
[INFO RSZ-0094] Found 3457 endpoints with setup violations.
[INFO RSZ-0045] Inserted 200 buffers, 1 to split loads.
[INFO RSZ-0041] Resized 195 instances.
[INFO RSZ-0043] Swapped pins on 80 instances.
[WARNING RSZ-0062] Unable to repair all setup violations.
[INFO RSZ-0033] No hold violations found.
Placement Analysis
---------------------------------
total displacement        985.8 u
average displacement        0.0 u
max displacement           11.0 u
original HPWL         5532893.1 u
legalized HPWL        5533517.5 u
delta HPWL                    0 %

**System 2 (2550 seconds):**

Repair setup and hold violations...
TNS end percent 100
Skipping gate cloning during optimization
[INFO RSZ-0094] Found 3459 endpoints with setup violations.
[INFO RSZ-0045] Inserted 539 buffers, 1 to split loads.
[INFO RSZ-0041] Resized 2955 instances.
[INFO RSZ-0043] Swapped pins on 118 instances.
[WARNING RSZ-0062] Unable to repair all setup violations.
[INFO RSZ-0033] No hold violations found.
Placement Analysis
---------------------------------
total displacement       6384.6 u
average displacement        0.0 u
max displacement           19.6 u
original HPWL         5567471.6 u
legalized HPWL        5572355.7 u
delta HPWL                    0 %

The difference i see here with respect to total displacement.

Please do let me know how to fix this.

### Expected Behavior

CTS should take  around 625 seconds in both systems. As that is nearer to golden value. 

### Environment

```shell
kernel: Linux 5.15.0-86-generic
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is Clang 16.0.3
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/amd/compiler/aocc4.1/aocc-compiler-rel-4.1.0-4451-270/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-10770-g6719c92e6
-- System name: Linux
-- Compiler: Clang 16.0.3
-- Build type: RELEASE
[... 126 lines cut ...]
-- Found OpenMP_CXX: -fopenmp=libomp (found version "5.0")
-- Found OpenMP: TRUE (found version "5.0")
-- GPU is not enabled
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0")
-- GUI is enabled
-- Charts widget is enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found version "1.83.0") found components: serialization
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS)
-- Found Boost: /usr/local/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.78")
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found version "1.83.0") found components: serialization system thread
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.8IHetRMv7r
```

### To Reproduce

This issue is specific to one of  the server i am working on.

QuantamHD, 2023-11-21:
Are the binaries on both systems byte for byte identical? Also do you have a test case that you can share with us?

maliberty, 2023-11-21:
What are the two systems?  You are getting different answers so something is varying beyond runtime.

sandeep-amd, 2023-11-21:
The binaries are not byte to byte identical. But both are built with same compiler(AOCC). Binary size is almost similar i.e OpenRoad is 79 MB on both systems.  Both are AMD servers with similar configuration .

maliberty, 2023-11-21:
Why not use the same binary to eliminate a source of difference?  Also is the input data identical?

tspyrou, 2023-11-22:
@sandeep-amd if you send an email to info@precisioninno.com, we can get connected and figure out how to solve this. -Tom

sandeep-amd, 2023-11-24:
Yes the input data is identical as it is being run using Open Road Flow Scripts. 

Also, if yoy see the log, major difference seems to be in this step.
[INFO RSZ-0045] Inserted 200 buffers, 1 to split loads.
[INFO RSZ-0041] Resized 195 instances.
[INFO RSZ-0043] Swapped pins on 80 instances.

#################################################################
[INFO RSZ-0045] Inserted 539 buffers, 1 to split loads.
[INFO RSZ-0041] Resized 2955 instances.
[INFO RSZ-0043] Swapped pins on 118 instances.
#################################################################

Why it has inserted more buffers(200 vs 539) and resized more instances (195 vs 2955) ? What does these steps do ?

sandeep-amd, 2023-11-24:
> @sandeep-amd if you send an email to [info@precisioninno.com](mailto:info@precisioninno.com), we can get connected and figure out how to solve this. -Tom

i will send an email

maliberty, 2023-11-24:
It is performing timing optimization.  You need to use the same build of OR to have comparable results.  The simplest way to ensure that would be to use the same binaries on both machines.

maliberty, 2024-03-05:
Were you able to connect with @tspyrou ?

tspyrou, 2024-03-11:
@sandeep-amd I don't think that we connected yet.