The-OpenROAD-Project/OpenROAD #4297: global route segfault in seconds

oharboe, 2023-11-22:
### Describe the bug

unzip  [grtcrash2.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/13434706/grtcrash2.zip)

```
$ ./run-me-regfile_128x65-asap7-base.sh 
OpenROAD v2.0-11036-gf6b82623c 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING STA-0337] port 'R0_clk' not found.
[WARNING STA-0357] virtual clock R0_clk can not be propagated.
[WARNING STA-0337] port 'W0_clk' not found.
[WARNING STA-0357] virtual clock W0_clk can not be propagated.
[WARNING STA-0337] port 'R0_addr[0]' not found.
[WARNING STA-0337] port 'R0_addr[1]' not found.
[WARNING STA-0337] port 'R0_addr[2]' not found.
[WARNING STA-0337] port 'R0_addr[3]' not found.
[WARNING STA-0337] port 'R0_addr[4]' not found.
[WARNING STA-0337] port 'R0_addr[5]' not found.
[WARNING STA-0337] port 'R0_addr[6]' not found.
[WARNING STA-0337] port 'R0_en' not found.
[WARNING STA-0337] port 'R1_addr[0]' not found.
[WARNING STA-0337] port 'R1_addr[1]' not found.
[WARNING STA-0337] port 'R1_addr[2]' not found.
[WARNING STA-0337] port 'R1_addr[3]' not found.
[WARNING STA-0337] port 'R1_addr[4]' not found.
[WARNING STA-0337] port 'R1_addr[5]' not found.
[WARNING STA-0337] port 'R1_addr[6]' not found.
[WARNING STA-0337] port 'R1_clk' not found.
[WARNING STA-0337] port 'R1_en' not found.
[WARNING STA-0337] port 'R2_addr[0]' not found.
[WARNING STA-0337] port 'R2_addr[1]' not found.
[WARNING STA-0337] port 'R2_addr[2]' not found.
[WARNING STA-0337] port 'R2_addr[3]' not found.
[WARNING STA-0337] port 'R2_addr[4]' not found.
[WARNING STA-0337] port 'R2_addr[5]' not found.
[WARNING STA-0337] port 'R2_addr[6]' not found.
[WARNING STA-0337] port 'R2_clk' not found.
[WARNING STA-0337] port 'R2_en' not found.
[WARNING STA-0337] port 'R3_addr[0]' not found.
[WARNING STA-0337] port 'R3_addr[1]' not found.
[WARNING STA-0337] port 'R3_addr[2]' not found.
[WARNING STA-0337] port 'R3_addr[3]' not found.
[WARNING STA-0337] port 'R3_addr[4]' not found.
[WARNING STA-0337] port 'R3_addr[5]' not found.
[WARNING STA-0337] port 'R3_addr[6]' not found.
[WARNING STA-0337] port 'R3_clk' not found.
[WARNING STA-0337] port 'R3_en' not found.
[WARNING STA-0337] port 'R4_addr[0]' not found.
[WARNING STA-0337] port 'R4_addr[1]' not found.
[WARNING STA-0337] port 'R4_addr[2]' not found.
[WARNING STA-0337] port 'R4_addr[3]' not found.
[WARNING STA-0337] port 'R4_addr[4]' not found.
[WARNING STA-0337] port 'R4_addr[5]' not found.
[WARNING STA-0337] port 'R4_addr[6]' not found.
[WARNING STA-0337] port 'R4_clk' not found.
[WARNING STA-0337] port 'R4_en' not found.
[WARNING STA-0337] port 'R5_addr[0]' not found.
[WARNING STA-0337] port 'R5_addr[1]' not found.
[WARNING STA-0337] port 'R5_addr[2]' not found.
[WARNING STA-0337] port 'R5_addr[3]' not found.
[WARNING STA-0337] port 'R5_addr[4]' not found.
[WARNING STA-0337] port 'R5_addr[5]' not found.
[WARNING STA-0337] port 'R5_addr[6]' not found.
[WARNING STA-0337] port 'R5_clk' not found.
[WARNING STA-0337] port 'R5_en' not found.
[WARNING STA-0337] port 'W0_addr[0]' not found.
[WARNING STA-0337] port 'W0_addr[1]' not found.
[WARNING STA-0337] port 'W0_addr[2]' not found.
[WARNING STA-0337] port 'W0_addr[3]' not found.
[... 169 lines cut ...]
13# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
14# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
15# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
16# ord::tclAppInit(Tcl_Interp*) in openroad
17# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
18# main in openroad
19# 0x00007F7F81823A90 in /lib/x86_64-linux-gnu/libc.so.6
20# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
21# _start in openroad
./run-me-regfile_128x65-asap7-base.sh: line 7: 94289 Aborted                 (core dumped) openroad -exit -no_init ${SCRIPTS_DIR}/global_route.tcl
```

### Expected Behavior

no crash

### Environment

```shell
OpenROAD v2.0-11036-gf6b82623c
```

### To Reproduce

see above

maliberty, 2023-11-22:
The stack suggests it is in find_ports_matching not global routing.  I wonder if you delta debugged down to a different issue?

oharboe, 2023-11-22:
> The stack suggests it is in find_ports_matching not global routing. I wonder if you delta debugged down to a different issue?

Yes... I think it is too good to be true that the 15 minute problem is now 10 seconds... It should be worth fixing though: fewer deltadebug false positives and shorter test-cases in the future...

maliberty, 2023-11-22:
Agreed, just wanted to point out the disconnect. I often delta with a function name from the stack

maliberty, 2023-11-22:
The reduced test case has two top level 'pins':
```
PINS 2 ;
    - W2_data[5] + NET W2_data[5] + DIRECTION INPUT + USE SIGNAL
    - W2_data[60] + NET W2_data[60] + DIRECTION INPUT + USE SIGNAL
END PINS
```
When we try to execute "get_ports {W2_data[10]}" from the SDC it causes a crash as the code in ConcretePort::findBusBit assumes all the bits are present in 

```
  else if (from_index_ >= to_index_
	   && index >= to_index_
	   && index <= from_index_)
    return (*member_ports_)[from_index_ - index];
```

maliberty, 2023-11-30:
Fixed by https://github.com/The-OpenROAD-Project/OpenROAD/pull/4331

closed 2023-11-30 by maliberty