The-OpenROAD-Project/OpenROAD #4298: cells overlap power grid in detailed placement

OuDret, 2023-11-22:
### Describe the bug

When running detailed placement placed cells overlap power stripes as shown in the following discussion [#4258](https://github.com/The-OpenROAD-Project/OpenROAD/discussions/4258). 

I'm using simplified nangate45 lef/def & lib files since i wanted to be sure that the problem was not in the pdk i was using.
By simplified i mean that i have removed unused cells and renamed cells to match verilog. I have also renamed some names like site.

### Expected Behavior

Placed cells does not align with power stripes

### Environment

```shell
Git commit: ec1c9f9450d49b49cb39b5b1381b42cf9bf9b800
kernel: Linux 6.2.0-37-generic
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-11153-gec1c9f945
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
[... 367 lines cut ...]
[INFO GPL-0072] 2.0rcK: 0.0
[INFO GPL-0073] 5.0rcK: 0.0
[INFO GPL-0074] FinalRC: 0.65000004
[NesterovSolve] Iter: 340 overflow: 0.209598 HPWL: 375579
[NesterovSolve] Iter: 350 overflow: 0.193679 HPWL: 374811
[NesterovSolve] Iter: 360 overflow: 0.176135 HPWL: 375655
[NesterovSolve] Iter: 370 overflow: 0.143264 HPWL: 377891
[NesterovSolve] Iter: 380 overflow: 0.116553 HPWL: 379052
[NesterovSolve] Iter: 390 overflow: 0.122842 HPWL: 381791
[NesterovSolve] Finished with Overflow: 0.099987
[WARNING RSZ-0014] wire capacitance for corner default is zero. Use the set_wire_rc command to set wire resistance and capacitance.
Placement Analysis
---------------------------------
total displacement         37.0 u
average displacement        0.6 u
max displacement            2.1 u
original HPWL             392.3 u
legalized HPWL            417.9 u
delta HPWL                    7 %
```

### Screenshots

[image]
[image]

maliberty, 2023-11-22:
Your sites/straps are incorrect which will mislead the placer.  The straps should align with the top/bottom of sites but you have straps going through the middle of cells.  Either your floorplan or pdn setup is incorrect.

OuDret, 2023-11-22:
Thank you, 
knowing where the error is makes it a lot easier to debug. I will close the issue.

closed 2023-11-22 by OuDret