The-OpenROAD-Project/OpenROAD #4299: [WARNING ODB-0177] error: undefined via (via5_6_120_288_1_2_58_322) referenced

oharboe, 2023-11-22:
### Describe the bug

unzip https://drive.google.com/file/d/1_A5dkVbY-6R_asjAVfNjOLVq0C0Eo3lc/view?usp=sharing

Run:

```
./run-me-BoomTile-asap7-base.sh
```

It takes a minute or so, and you find this error.  Look for `data_2048x2.lef ` specifically, because I will provide more detail on it below.

```
[deleted]
[INFO ODB-0222] Reading LEF file: bazel-bin/build/results/asap7/data_2048x2/base/data_2048x2.lef
[WARNING ODB-0177] error: undefined via (via5_6_120_288_1_2_58_322) referenced
[WARNING ODB-0177] error: undefined via (via5_6_120_288_1_2_58_322) referenced
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  bazel-bin/build/results/asap7/data_2048x2/base/data_2048x2.lef
[deleted]
```

unzip https://drive.google.com/file/d/1cNIRERAaOmR2GFtT3EOxc2kx1v91uaJ5/view?usp=sharing generated with  `make generate_abstract_issue`, which will generate `data_2048x2.lef`:

```
./run-me-data_2048x2-asap7-base.sh
```

```
OpenROAD v2.0-11072-g2d83b7137 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Starting generation of abstract views
```

If you need to generate the `data_2048x2.lef` from scratch, check out https://github.com/hdl/bazel_rules_hdl/pull/225 and run:

```
cd megaboom/
./orfs make WORK_HOME=build  DESIGN_NAME=data_2048x2 SDC_FILE=constraints-sram.sdc VERILOG_FILES=rtl/[data_2048x2.sv](http://data_2048x2.sv/) PLACE_DENSITY=0.65 generate_abstract
```

### Expected Behavior

No error

### Environment

```shell
OpenROAD v2.0-11072-g2d83b7137
```

### To Reproduce

See above

oharboe, 2023-11-22:
@maliberty Please confirm that this issue contains sufficient information to reproduce the problem?

maliberty, 2023-11-23:
I can reproduce it.  The LEF abstract writer needs to include the generated vias.

closed 2023-11-30 by maliberty