The-OpenROAD-Project/OpenROAD #4301: [WARNING PPL-0015] Macro core/FpPipeline/fregfile/regfile_ext is not placed.

oharboe, 2023-11-23:
### Describe the bug

unzip https://drive.google.com/file/d/17-H3xSv_emeozTsXrW7eKzUb2y-uAjYS/view?usp=sharing

```
$ ./run-me-BoomTile-asap7-base.sh 
OpenROAD v2.0-11072-g2d83b7137 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING PPL-0015] Macro core/FpPipeline/fregfile/regfile_ext is not placed.
[WARNING PPL-0015] Macro core/iregfile/regfile_ext is not placed.
[WARNING PPL-0015] Macro dcache/data/array_0_0_0_ext is not placed.
[WARNING PPL-0015] Macro dcache/data/array_0_0_1_ext is not placed.
[WARNING PPL-0015] Macro dcache/data/array_0_1_0_ext is not placed.
[WARNING PPL-0015] Macro dcache/data/array_0_1_1_ext is not placed.
[deleted]
```

### Expected Behavior

No warning

### Environment

```shell
OpenROAD v2.0-11072-g2d83b7137
```

### To Reproduce

See above

### Additional Context

This is from the same test project as for which #4299 was filed, but possibly a separate problem.

maliberty, 2023-11-23:
Are you not running macro placement before pin placement?

oharboe, 2023-11-23:
> Are you not running macro placement before pin placement?

I'm using ORFS and doing whatever ORFS is doing...

The command I ran before running `make io_placement_random_issue` was:

```
./orfs make do-floorplan
```

It runs macro placement *after* io_placement_random.tcl:

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/c72ffd7dcd317c337df2535314289f8d0a593467/flow/Makefile#L609

oharboe, 2023-11-23:
@maliberty Perhaps worth fixing #4299 and then retesting?

oharboe, 2023-11-23:
@maliberty Tried with the #4302 fix, still problems.

unzip [io_placement_random.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/13451004/io_placement_random.zip)

```
$ ./run-me-L1MetadataArray-asap7-base.sh 
OpenROAD v2.0-11160-g042cdcfa4 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING PPL-0015] Macro tag_array_0_ext is not placed.
[WARNING PPL-0015] Macro tag_array_1_ext is not placed.
[WARNING PPL-0015] Macro tag_array_2_ext is not placed.
[WARNING PPL-0015] Macro tag_array_3_ext is not placed.
[WARNING PPL-0015] Macro tag_array_4_ext is not placed.
[WARNING PPL-0015] Macro tag_array_5_ext is not placed.
[WARNING PPL-0015] Macro tag_array_6_ext is not placed.
[WARNING PPL-0015] Macro tag_array_7_ext is not placed.
Found 0 macro blocks.
Using 2 tracks default min distance between IO pins.
[INFO PPL-0007] Random pin placement.
openroad> 
```

eder-matheus, 2023-11-23:
@oharboe In ORFS, the random pin placement runs before the macro placement, so this is expected. We need an initial pin placement for the following steps, so we run random before everything.

I could update PPL to avoid these warning messages, but it is not necessarily an error since it is expected.

oharboe, 2023-11-23:
@eder-matheus Aha! So it is a bogus warning?

The macros are visible at the end of the floorplan...

[image]

oharboe, 2023-11-23:
@eder-matheus So this github issue is really a feature request: remove bogus warning :-)

eder-matheus, 2023-11-23:
It isn't a bogus warning because it doesn't have the macro placement when PPL runs. You'll see the macros at the end of the floorplan stage because the macro placement runs after PPL.

But I think the best solution would be to remove the warning message when we run random PPL since it will not look at the placement of any macro or std cell.

closed 2023-11-23 by maliberty