The-OpenROAD-Project/OpenROAD #4306: floorplan.tcl does not identify macros - [WARNING ODB-0208] VIA: duplicate VIA (via5_6_120_288_1_2_58_322) ignored...

oharboe, 2023-11-23:
### Describe the bug

After #4302 was merged, I still get this warning below:

To reproduce unzip https://drive.google.com/file/d/1ymXux_59zURs7Fh35NxYGcnBvkOe8YeN/view?usp=sharing

```
./run-me-BoomTile-asap7-base.sh
```

Output:

```
OpenROAD v2.0-11160-g0728be75e
[deleted]
[WARNING ODB-0208] VIA: duplicate VIA (via5_6_120_288_1_2_58_322) ignored...
[deleted]
```

### Expected Behavior

no error

### Environment

```shell
OpenROAD v2.0-11160-g0728be75e
```

### To Reproduce

See above.

maliberty, 2023-11-24:
I suppose this is because you have multiple block abstracts with the same via.  We could lower it to an info.

oharboe, 2023-11-24:
> I suppose this is because you have multiple block abstracts with the same via. We could lower it to an info.

What will the user do with this information?

Who needs to know and why?

rovinski, 2023-11-29:
A user could have typos or copy/paste errors. Vias that were meant to be different could accidentally have the same name.

oharboe, 2023-12-13:
@maliberty I no longer see it, close?

closed 2023-12-13 by maliberty