The-OpenROAD-Project/OpenROAD #4311: Unable to build with newer spdlog (fmt v10)

CircuitCoder, 2023-11-25:
### Describe the bug

When building with newer spdlog which uses fmt v10, the build fails with:
```
In file included from /usr/include/spdlog/fmt/fmt.h:31,
                 from /root/workspace/CRAFT/OpenROAD-flow-scripts/tools/OpenROAD/src/utl/include/utl/Logger.h:52,
                 from /root/workspace/CRAFT/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/db/dbBTerm.cpp:59:
/usr/include/fmt/core.h: In instantiation of ‘constexpr fmt::v10::detail::value<Context> fmt::v10::detail::make_arg(T&) [with bool PACKED = true; Context = fmt::v10::basic_format_context<fmt::v10::appender, char>; T = const odb::dbSigType::Value; typename std::enable_if<PACKED, int>::type <anonymous> = 0]’:
/usr/include/fmt/core.h:1808:51:   required from ‘constexpr fmt::v10::format_arg_store<Context, Args>::format_arg_store(T& ...) [with T = {const char* const, const char* const, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, const odb::dbSigType::Value}; Context = fmt::v10::basic_format_context<fmt::v10::appender, char>; Args = {const char*, const char*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, odb::dbSigType::Value}]’
/usr/include/fmt/core.h:1826:18:   required from ‘constexpr fmt::v10::format_arg_store<Context, typename std::remove_cv<typename std::remove_reference<T>::type>::type ...> fmt::v10::make_format_args(T& ...) [with Context = basic_format_context<appender, char>; T = {const char* const, const char* const, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, const odb::dbSigType::Value}]’
/usr/include/spdlog/logger.h:374:75:   required from ‘void spdlog::logger::log_(spdlog::source_loc, spdlog::level::level_enum, spdlog::string_view_t, Args&& ...) [with Args = {const char* const&, const char* const&, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, const odb::dbSigType::Value&}; spdlog::string_view_t = fmt::v10::basic_string_view<char>]’
/usr/include/spdlog/logger.h:90:13:   required from ‘void spdlog::logger::log(spdlog::source_loc, spdlog::level::level_enum, fmt::v10::format_string<T ...>, Args&& ...) [with Args = {const char* const&, const char* const&, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, const odb::dbSigType::Value&}; fmt::v10::format_string<T ...> = fmt::v10::basic_format_string<char, const char* const&, const char* const&, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, const odb::dbSigType::Value&>]’
/usr/include/spdlog/logger.h:96:12:   required from ‘void spdlog::logger::log(spdlog::level::level_enum, fmt::v10::format_string<T ...>, Args&& ...) [with Args = {const char* const&, const char* const&, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, const odb::dbSigType::Value&}; fmt::v10::format_string<T ...> = fmt::v10::basic_format_string<char, const char* const&, const char* const&, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, const odb::dbSigType::Value&>]’
/root/workspace/CRAFT/OpenROAD-flow-scripts/tools/OpenROAD/src/utl/include/utl/Logger.h:131:17:   required from ‘void utl::Logger::debug(utl::ToolId, const std::string&, const std::string&, const Args& ...) [with Args = {odb::dbSigType::Value}; std::string = std::__cxx11::basic_string<char>]’
/root/workspace/CRAFT/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/db/dbBTerm.cpp:332:5:   required from here
/usr/include/fmt/core.h:1576:63: error: ‘fmt::v10::detail::type_is_unformattable_for<const odb::dbSigType::Value, char> _’ has incomplete type
 1576 |     type_is_unformattable_for<T, typename Context::char_type> _;
      |                                                               ^
/usr/include/fmt/core.h:1580:7: error: static assertion failed: Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt
 1580 |       formattable,
      |       ^~~~~~~~~~~
/usr/include/fmt/core.h:1580:7: note: ‘formattable’ evaluates to false
/usr/include/fmt/core.h: In instantiation of ‘constexpr fmt::v10::detail::value<Context> fmt::v10::detail::make_arg(T&) [with bool PACKED = true; Context = fmt::v10::basic_format_context<fmt::v10::appender, char>; T = const odb::dbIoType::Value; typename std::enable_if<PACKED, int>::type <anonymous> = 0]’:
/usr/include/fmt/core.h:1808:51:   required from ‘constexpr fmt::v10::format_arg_store<Context, Args>::format_arg_store(T& ...) [with T = {const char* const, const char* const, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, const odb::dbIoType::Value}; Context = fmt::v10::basic_format_context<fmt::v10::appender, char>; Args = {const char*, const char*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, odb::dbIoType::Value}]’
/usr/include/fmt/core.h:1826:18:   required from ‘constexpr fmt::v10::format_arg_store<Context, typename std::remove_cv<typename std::remove_reference<T>::type>::type ...> fmt::v10::make_format_args(T& ...) [with Context = basic_format_context<appender, char>; T = {const char* const, const char* const, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, const odb::dbIoType::Value}]’
/usr/include/spdlog/logger.h:374:75:   required from ‘void spdlog::logger::log_(spdlog::source_loc, spdlog::level::level_enum, spdlog::string_view_t, Args&& ...) [with Args = {const char* const&, const char* const&, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, const odb::dbIoType::Value&}; spdlog::string_view_t = fmt::v10::basic_string_view<char>]’
/usr/include/spdlog/logger.h:90:13:   required from ‘void spdlog::logger::log(spdlog::source_loc, spdlog::level::level_enum, fmt::v10::format_string<T ...>, Args&& ...) [with Args = {const char* const&, const char* const&, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, const odb::dbIoType::Value&}; fmt::v10::format_string<T ...> = fmt::v10::basic_format_string<char, const char* const&, const char* const&, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, const odb::dbIoType::Value&>]’
/usr/include/spdlog/logger.h:96:12:   required from ‘void spdlog::logger::log(spdlog::level::level_enum, fmt::v10::format_string<T ...>, Args&& ...) [with Args = {const char* const&, const char* const&, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, const odb::dbIoType::Value&}; fmt::v10::format_string<T ...> = fmt::v10::basic_format_string<char, const char* const&, const char* const&, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, const odb::dbIoType::Value&>]’
/root/workspace/CRAFT/OpenROAD-flow-scripts/tools/OpenROAD/src/utl/include/utl/Logger.h:131:17:   required from ‘void utl::Logger::debug(utl::ToolId, const std::string&, const std::string&, const Args& ...) [with Args = {odb::dbIoType::Value}; std::string = std::__cxx11::basic_string<char>]’
/root/workspace/CRAFT/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/src/db/dbBTerm.cpp:358:5:   required from here
/usr/include/fmt/core.h:1576:63: error: ‘fmt::v10::detail::type_is_unformattable_for<const odb::dbIoType::Value, char> _’ has incomplete type
 1576 |     type_is_unformattable_for<T, typename Context::char_type> _;
      |                                                               ^
/usr/include/fmt/core.h:1580:7: error: static assertion failed: Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt
 1580 |       formattable,
      |       ^~~~~~~~~~~
/usr/include/fmt/core.h:1580:7: note: ‘formattable’ evaluates to false
```

This issues seems to have previously happened when updating to fmt v9 (#3228), and fmt v10 just removed `FMT_DEPRECATED_OSTREAM` (https://github.com/fmtlib/fmt/releases/tag/10.0.0), so the workaround is not working.

### Expected Behavior

OpenROAD should be able to be built against fmt v10.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.133.1-microsoft-standard-WSL2
os: Arch Linux
cmake version 3.27.8
[... cut ...]
./src/odb/src/db/dbStream.cpp:std::ostream& operator<<(std::ostream& os, const Orientation2D& ori)
./src/odb/src/db/dbStream.cpp:std::ostream& operator<<(std::ostream& os, const Orientation3D& ori)
./src/odb/src/db/dbStream.cpp:std::ostream& operator<<(std::ostream& os, const Direction1D& dir)
./src/odb/src/db/dbStream.cpp:std::ostream& operator<<(std::ostream& os, const Direction2D& dir)
./src/odb/src/db/dbStream.cpp:std::ostream& operator<<(std::ostream& os, const Direction3D& dir)
```

I'm trying to convert them to fmt formatters to see if it can make OpenROAD compile.

CircuitCoder, 2023-11-26:
> I think the best solution is FMT_DEPRECATED_OSTREAM as discussed in #2696

Unfortunately fmtv10 just removed that switch. I'm compiling with spdlog v1.12.0 and fmt v10.1.1 on ArchLinux. 

AFAIK not a lot of other distros have fmt v10. Fedora 39 has it, and Debian experimental has it. But I assume it will become a problem later on.

maliberty, 2023-11-26:
https://github.com/fmtlib/fmt/issues/3088 is a solution that they proposed but I haven't had a chance to try it.  Perhaps you could take a look?

rovinski, 2023-12-08:
Resolved?