The-OpenROAD-Project/OpenROAD #4318: GPL padding cause increase in reported DU

lustefan97, 2023-11-27:
### Describe the bug

Hello,

I have an issue when trying to increase the target design utilization. Whenever I increase it to or above 70%, GPL reports a DU that is way above what I am aiming for or even in this case superior to 100%, which causes the tool to stop.

The issue comes from the padding in GPL. Whenever I set it to 0 (`set global_place_pad 2` & `set detail_place_pad 1` in asap7_bottom.vars) then the proper DU is reported, I don't know if this is the intended behavior, but I feel like space padding present existing to facilitate routing should not influence the DU...

### Expected Behavior

Maybe some space padding, but no influence on the reported design utilization value

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.1.0-11-amd64
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-11117-g6061c4892
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
[... 78 lines cut ...]
-- GPU is not enabled
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.1") 
-- GUI is enabled
-- Charts widget is not enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.N0C0Gj9QSZ
```

### To Reproduce

[Uploading TestCase.tar.gz…]()

Unzip this directory inside _OpenROAD-flow-scripts/tools/OpenROAD/test_ (make sure that everything is inside the TestCase directory), and use `bash run_ibex_cst_pd.sh`. The script should start.

lustefan97, 2023-11-30:
@rovinski But wouldn't that reduce the maximum design utilization possible for a design ? Especially since OpenROAD want to be automatic, GPL will stop everything if it sees a DU >100% even if it could be possible to achieve a reasonable DU with an equivalent padding ?

I encountered this "issue" when trying to aim for a DU of 70% on the ASAP7 PDK which should be possible and reasonable for any design

rovinski, 2023-12-01:
> But wouldn't that reduce the maximum design utilization possible for a design ?

Yes, that's exactly the point. If you don't want padding in your design, then you should reduce it or turn it off. But be forewarned that padding is usually necessary to achieve a cleanly routed design. If you want to optimize for very high density designs, I believe you can also set padding on a master-by-master basis so that only the most difficult cells are padded. That would take some trial and error.

> when trying to aim for a DU of 70%

Is this the core utilization after synthesis? Because after timing optimization the utilization can also increase dramatically due to gate resizing and buffer insertion.

lustefan97, 2023-12-01:
By after synthesis do you mean after the synthesis step in Yosys, or after the PnR step ?

To me the synthesis step only convert the verilog code into a gate netlist mapped to the target pdk, it gets no information about design utilization yet ? But yes after PnR I target 70%, and GPL (with cell padding) reports a DU above 100%. I am perfectly fine with having a DU a bit above or below that value, as long as it is around it.

rovinski, 2023-12-01:
If you are using ORFS, the `CORE_UTILIZATION` (which is what I assume you are referring to) is actually not that well defined and usually isn't an indicator of the utilization you end up with.

After Yosys (synthesis) runs, you have a gate-level netlist. During floorplanning, the design area from this unmodified netlist is used to calculate how big the die and core sizes should be (design area / `CORE_UTILIZATION` = `CORE_AREA`). After it does that, the die and core areas are fixed.

The utilization will immediately go down because ORFS removes all buffers inserted by Yosys. The utilization will then go up again, usually by a lot, because OpenROAD does rebuffering and gate resizing in order to meet timing constraints. Timing constraints are critical and the algorithm does not factor in utilization targets at all. You could easily start with a design at 70% utilization during the initialization step, and then go up to 90% because of optimization. Once you add padding during GPL, utilization can exceed 100%.

Why aren't the tools smarter? Why doesn't it account for this area increase due to timing optimization? Well because you have to do the entire timing optimization to figure out how much area it will take up. Trying to estimate this is an active area of research.

Everything is a trial and error process. If for some reason you specifically need a 70% utilization, then I would suggest you extremely relax your timing constraints so that the optimizer doesn't need to add much. Otherwise, you should simply start with a lower utilization, allow timing optimization to do its thing, and then finish with approximately whatever target utilization you're after.

lustefan97, 2023-12-04:
So, if I understood correctly the timing optimizations done by the synthesis tool are "discarded" as the buffers are removed, then replaced during PnR where the DU is also influenced by the timing constraints ?

So, for instance, if I have a design for which the detailed placement fails for a few instances, if I relax my timing constraints it may result in a successful placement ?

rovinski, 2023-12-04:
> the timing optimizations done by the synthesis tool are "discarded" as the buffers are removed

Yes. We have found that OpenROAD optimizes better when the buffers from Yosys are removed, because Yosys does not do physical synthesis.

> if I have a design for which the detailed placement fails for a few instances, if I relax my timing constraints it may result in a successful placement ?

Yes. Relaxed timing constraints usually translates to less buffer insertion and smaller design area.

lustefan97, 2023-12-05:
Thanks for your help ! I do have another question, does this means that there no need to re-synthesize with different timing constraint with Yosys ? Or does it also use different/bigger non-buffer cells that also need to be resized ?

maliberty, 2023-12-05:
You may get a different result with different timing constraints as gates may be sized, not only buffers.

rovinski, 2023-12-05:
To add, historically we have not seen much change in the output of Yosys based on the timing constraints provided, but the results aren't necessarily identical.

lustefan97, 2023-12-06:
Ok, I understand what is happening a bit better, I'll close this issue. Thank you very much for your explanations @rovinski !

closed 2023-12-06 by lustefan97