The-OpenROAD-Project/OpenROAD #4319: GRT warning lead to DRT error : Guide is not connected to design

lustefan97, 2023-11-27:
### Describe the bug

I am trying to run a custom script to characterize OpenROAD's behavior when the design utilization is varying for a fixed placement_density, using the ASAP7 PDK, when all IO's are on the top metal layer.

I currently use the ibex design as a test design and I am aiming at a DU of 60%, with a placement density fixed at 1. The issue I have is that I encounter this warning during the global routing step :

"[WARNING GRT-0026] Missing route to pin (name of the pin)"

It does this for all pins. During detailed routing it seems to try and fix this problem (as OpenROAD give me the [INFO-DRT-1000] code), but it fails to do so and thus stop the flow after indicating :

"[ERROR DRT-0218] Guide is not connected to design"

I want to fix this issue, but I have no idea as to where to look, or what to do...

I use the "grid-strategy-M1-M2-M5-M6.tcl" as well as the make_tracks.tcl found in the OpenROAD-flow-scripts/flow/platforms/asap7/ directory.

### Expected Behavior

No warning or error raised

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.1.0-11-amd64
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-11117-g6061c4892
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
[... 88 lines cut ...]
-- TCL header: /usr/include/tcl/tcl.h
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.1") 
-- GUI is enabled
-- Charts widget is not enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.N0C0Gj9QSZ

(The version I use has been installed on Friday 11/24)
```

### To Reproduce

[Uploading TestCase.tar.gz…]()

Here is a test case, uncompress the file into _OpenROAD-flow-scripts/tools/OpenROAD/test/_, enter the TestCase directory and use `bash run_ibex_cst_pd.sh` to reproduce the issue

eder-matheus, 2023-11-27:
@lustefan97 The test case link is incorrect, it doesn't download anything.

lustefan97, 2023-11-27:
@eder-matheus I don't know what happened but I updated the link, tell me if you can download it now

closed 2023-11-29 by eder-matheus

eder-matheus, 2023-11-29:
@lustefan97 The fix for it is in the master branch. Could you confirm it is working for you?

lustefan97, 2023-11-30:
@eder-matheus Yes it does, thanks for your help ! I do have another warning that I had on another case that I don't understand though. It's about some nets (I think for output pins) that are said not connected to any iterm or bterm yet when looking in the gui it seems like it is the case. 

I'll open another issue with a test case for you to check out