The-OpenROAD-Project/OpenROAD #4321: Improve error message: [ERROR GPL-0307] RePlAce divergence detected. Re-run with a smaller max_phi_cof value.

oharboe, 2023-11-28:
### Description

What does ORFS / OpenROAD want me to do here?

```
[NesterovSolve] Iter: 410 overflow: 0.198856 HPWL: 39454131571
[NesterovSolve] Iter: 420 overflow: 0.191851 HPWL: 41732136463
[NesterovSolve] Iter: 430 overflow: 0.173834 HPWL: 50031597863
[NesterovSolve] Iter: 440 overflow: 0.162364 HPWL: 54937616026
[ERROR GPL-0307] RePlAce divergence detected. Re-run with a smaller max_phi_cof value.
Error: tdms_place.tcl, 28 GPL-0307
Command exited with non-zero status 1
Elapsed time: 1:02:01[h:]min:sec. CPU time: user 7652.95 sys 30.82 (206%). Peak memory: 13124976KB.
```

### Suggested Solution

Change the message to be something actionable by the user

maliberty, 2023-11-28:
I agree it isn't a great message.  global_place has a -max_phi_coef argument but it is quite obscure in its effect.  You can try playing with it (it defaults to 1.05).

Usually what I find is that there is something else interesting going on that is preventing convergence.  Can you make a test case for this?

oharboe, 2023-11-29:
To reproduce, unzip https://drive.google.com/file/d/1fYQPW0ls_CkZ2lAWY9433JLV1Q7pWbcH/view?usp=sharing

Run as below. Note! Takes a while... 20 minutes or so?

```
$ ./run-me-BoomTile-asap7-base.sh 
OpenROAD v2.0-11183-g2f133cf95 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO GPL-0002] DBU: 1000
[INFO GPL-0003] SiteSize: 54 270
[INFO GPL-0004] CoreAreaLxLy: 2052 2160
[deleted]
[NesterovSolve] Iter: 420 overflow: 0.191851 HPWL: 41732136463
[NesterovSolve] Iter: 430 overflow: 0.173834 HPWL: 50031597863
[NesterovSolve] Iter: 440 overflow: 0.162364 HPWL: 54937616026
[ERROR GPL-0307] RePlAce divergence detected. Re-run with a smaller max_phi_cof value.
Error: tdms_place.tcl, 28 GPL-0307
```