The-OpenROAD-Project/OpenROAD #4334: CTS is verbose, move more output to debug level

oharboe, 2023-12-01:
### Description

```
$ ./run-me-BoomTile-asap7-base.sh 
OpenROAD v2.0-11160-g042cdcfa4 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
clock_tree_synthesis -root_buf BUFx4_ASAP7_75t_R -buf_list BUFx4_ASAP7_75t_R -sink_clustering_enable -sink_clustering_size 30 -sink_clustering_max_diameter 100 -balance_levels -distance_between_buffers 60
[INFO CTS-0049] Characterization buffer is: BUFx4_ASAP7_75t_R.
[INFO CTS-0039] Number of created patterns = 12240.
[INFO CTS-0084] Compiling LUT.
Min. len    Max. len    Min. cap    Max. cap    Min. slew   Max. slew
2           8           1           34          1           12          
[WARNING CTS-0043] 1632 wires are pure wire and no slew degradation.
TritonCTS forced slew degradation on these wires.
[INFO CTS-0046]     Number of wire segments: 12240.
[INFO CTS-0047]     Number of keys in characterization LUT: 1456.
[INFO CTS-0048]     Actual min input cap: 1.
[INFO CTS-0007] Net "clock" found for clock "clock".
[INFO CTS-0010]  Clock net "clock" has 352871 sinks.
[INFO CTS-0008] TritonCTS found 1 clock nets.
[INFO CTS-0097] Characterization used 1 buffer(s) types.
[INFO CTS-0200] 8 placement blockages have been identified.
[INFO CTS-0027] Generating H-Tree topology for net clock.
[INFO CTS-0028]  Total number of sinks: 352871.
[INFO CTS-0029]  Sinks will be clustered in groups of up to 30 and with maximum cluster diameter of 100.0 um.
[INFO CTS-0030]  Number of static layers: 0.
[INFO CTS-0020]  Wire segment unit: 1350  dbu (1 um).
[INFO CTS-0021]  Distance between buffers: 22 units (60 um).
[INFO CTS-0019]  Total number of sinks after clustering: 11866.
[INFO CTS-0024]  Normalized sink region: [(16.3511, 16.1382), (3524.09, 3510.48)].
[INFO CTS-0025]     Width:  3507.7341.
[INFO CTS-0026]     Height: 3494.3418.
 Level 1
    Direction: Horizontal
    Sinks per sub-region: 5933
    Sub-region size: 1753.8670 X 3494.3418
[INFO CTS-0034]     Segment length (rounded): 876.
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 264 inSlew: 1 inCap: 1 outSlew: 1 load: 1 length: 8 delay: 6
      location: 0.5 buffer: BUFx4_ASAP7_75t_R
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 336 inSlew: 1 inCap: 1 outSlew: 1 load: 1 length: 8 delay: 6
      location: 1.0 buffer: BUFx4_ASAP7_75t_R
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 264 inSlew: 1 inCap: 1 outSlew: 1 load: 1 length: 8 delay: 6
      location: 0.5 buffer: BUFx4_ASAP7_75t_R
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 336 inSlew: 1 inCap: 1 outSlew: 1 load: 1 length: 8 delay: 6
      location: 1.0 buffer: BUFx4_ASAP7_75t_R
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 348 inSlew: 1 inCap: 1 outSlew: 2 load: 1 length: 8 delay: 1
    Key: 264 inSlew: 1 inCap: 1 outSlew: 1 load: 1 length: 8 delay: 6
[... 721 lines cut ...]
[INFO CTS-0100]  Leaf buffers 11846
[INFO CTS-0101]  Average sink wire length 8833.40 um
[INFO CTS-0102]  Path depth 78 - 79
[WARNING STA-0357] virtual clock clock_vir can not be propagated.
Warning: There are 380 unconstrained endpoints.

==========================================================================
cts pre-repair report_check_types -max_slew -max_cap -max_fanout -violators
--------------------------------------------------------------------------
-148.0156
-1.0903

==========================================================================
cts pre-repair report_design_area
--------------------------------------------------------------------------
Design area 4644608 u^2 20% utilization.
[INFO RSZ-0058] Using max wire length 162um.
[INFO RSZ-0047] Found 79 long wires.
[INFO RSZ-0048] Inserted 295 buffers in 79 nets.
Warning: There are 380 unconstrained endpoints.
```

### Suggested Solution

Considering log above, change as approperiate to debug level output.

oharboe, 2023-12-20:
@maliberty This is a much more most recent output for CTS. Much less verbose.

Some comments on log quality and detail...

```
[INFO CTS-0050] Root buffer is BUFx24_ASAP7_75t_R.
[INFO CTS-0051] Sink buffer is BUFx24_ASAP7_75t_R.
[INFO CTS-0052] The following clock buffers will be used for CTS:
                    BUFx24_ASAP7_75t_R
[INFO CTS-0049] Characterization buffer is BUFx24_ASAP7_75t_R.
```

More or less the same information above repeated four times. Single line? Some of this superfluous?

```
[INFO CTS-0039] Number of created patterns = 4200.
```

What is a "pattern" and what conclusions can the user make from this information?

```
[INFO CTS-0084] Compiling LUT.
Min. len    Max. len    Min. cap    Max. cap    Min. slew   Max. slew
2           8           1           20          1           7           
```

Isn't LUT an implementation detail?

```
[WARNING CTS-0043] 560 wires are pure wire and no slew degradation.
TritonCTS forced slew degradation on these wires.
```

This warning is in the middle of the run and can only be found by grepping... If this is critical information, it would be easier to find at the bottom of the CTS run?

```
[INFO CTS-0046]     Number of wire segments: 2611.
[INFO CTS-0047]     Number of keys in characterization LUT: 305.
[INFO CTS-0048]     Actual min input cap: 1.
[INFO CTS-0007] Net "clock" found for clock "clock".
[INFO CTS-0010]  Clock net "clock" has 4 sinks.
[INFO CTS-0008] TritonCTS found 1 clock nets.
[INFO CTS-0097] Characterization used 1 buffer(s) types.
[INFO CTS-0200] 65 placement blockages have been identified.
[INFO CTS-0027] Generating H-Tree topology for net clock.
[INFO CTS-0028]  Total number of sinks: 4.
[INFO CTS-0090]  Sinks will be clustered based on buffer max cap.
[INFO CTS-0030]  Number of static layers: 0.
[INFO CTS-0020]  Wire segment unit: 1350  dbu (1 um).
[INFO CTS-0021]  Distance between buffers: 22 units (60 um).
[INFO CTS-0023]  Original sink region: [(543253, 1483741), (543523, 1486441)].
[INFO CTS-0024]  Normalized sink region: [(402.41, 1099.07), (402.61, 1101.07)].
[INFO CTS-0025]     Width:  0.2000.
[INFO CTS-0026]     Height: 2.0000.
```

There is a lot of information above... Hard to know what I should try to understand and what I can do about it...

```
[WARNING CTS-0045] Creating fake entries in the LUT.
 Level 1
    Direction: Vertical
    Sinks per sub-region: 2
    Sub-region size: 0.2000 X 1.0000
[INFO CTS-0034]     Segment length (rounded): 1.
[INFO CTS-0032]  Stop criterion found. Max number of sinks is 15.
[INFO CTS-0035]  Number of sinks covered: 4.
[INFO CTS-0018]     Created 3 clock buffers.
[INFO CTS-0012]     Minimum number of buffers in the clock path: 2.
[INFO CTS-0013]     Maximum number of buffers in the clock path: 2.
[INFO CTS-0015]     Created 3 clock nets.
[INFO CTS-0016]     Fanout distribution for the current clock = 2:2..
[INFO CTS-0017]     Max level of the clock tree: 1.
[INFO CTS-0098] Clock net "clock"
[INFO CTS-0099]  Sinks 4
[INFO CTS-0100]  Leaf buffers 0
[INFO CTS-0101]  Average sink wire length 159.41 um
[INFO CTS-0102]  Path depth 2 - 2
```

Hmm... Lots of information...

```
[WARNING STA-0357] virtual clock clock_vir can not be propagated.
```

The warning above is a bit drowned out and repeated many times in the logs due to the clock_vir that I have in my .sdc file.

```
Warning: There are 135660 unclocked register/latch pins.
```

What are a "unclocked register/latch" pins? There are no latches in this design.

```
Warning: There are 135902 unconstrained endpoints.
```

The .sdc file needs work...

```
==========================================================================
cts pre-repair report_design_area
--------------------------------------------------------------------------
Design area 1364898 u^2 51% utilization.
[INFO RSZ-0058] Using max wire length 162um.
```

```
Warning: There are 135660 unclocked register/latch pins.
Warning: There are 135902 unconstrained endpoints.
```

Repetition above...

```
==========================================================================
cts post-repair report_design_area
--------------------------------------------------------------------------
Design area 1364898 u^2 51% utilization.
Placement Analysis
---------------------------------
total displacement          1.0 u
average displacement        0.0 u
max displacement            0.3 u
```

What is "displacement"?

```
original HPWL        19804342.9 u
legalized HPWL       19948597.3 u
delta HPWL                    1 %

Repair setup and hold violations...
TNS end percent 5
[INFO RSZ-0033] No hold violations found.
Placement Analysis
---------------------------------
total displacement          0.0 u
average displacement        0.0 u
max displacement            0.0 u
original HPWL        19948597.3 u
legalized HPWL       19948597.3 u
delta HPWL                    0 %
```

```
Warning: There are 135660 unclocked register/latch pins.
Warning: There are 135902 unconstrained endpoints.
```

More repetition...

```
==========================================================================
cts final report_design_area
--------------------------------------------------------------------------
Design area 1364898 u^2 51% utilization.
Elapsed time: 46:23.04[h:]min:sec. CPU time: user 2769.95 sys 12.20 (99%). Peak memory: 17120504KB.
```

maliberty, 2023-12-21:
> More or less the same information above repeated four times. Single line? Some of this superfluous?

The various buffers happen to be the same here but they needn't be.  The messages describe different information.

> What is a "pattern" and what conclusions can the user make from this information?

Not much beyond more patterns == more runtime.  It can be moved to debug.

> Isn't LUT an implementation detail?

Yes, can be a debug.

> There is a lot of information above... Hard to know what I should try to understand and what I can do about it...

Mostly should be debug.

> What are a "unclocked register/latch" pins? There are no latches in this design.

It means registers or latch pins.  Unclocked means that they have no clock signal reaching their clock pin so they will get no timing check.

> What is "displacement"?

The distance an instance is moved when legalizing its location.

precisionmoon, 2023-12-30:
More CTS messages have been moved to debug in [PR 4459.](https://github.com/The-OpenROAD-Project/OpenROAD/pull/4459)

oharboe, 2024-01-12:
@maliberty @precisionmoon Is this done for now? There is still a lot of output I don't understand :-) and can't do anything about...

ChipTop now has ca. 250 lines of output. That's not an onerous task to wade through...

```
OpenROAD v2.0-11780-g3a01a779d 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
clock_tree_synthesis -sink_clustering_enable -balance_levels -distance_between_buffers 60
[INFO CTS-0050] Root buffer is BUFx24_ASAP7_75t_R.
[INFO CTS-0051] Sink buffer is BUFx24_ASAP7_75t_R.
[INFO CTS-0052] The following clock buffers will be used for CTS:
                    BUFx24_ASAP7_75t_R
[INFO CTS-0049] Characterization buffer is BUFx24_ASAP7_75t_R.
[INFO CTS-0007] Net "clock_uncore" found for clock "clock_uncore".
[INFO CTS-0010]  Clock net "clock_uncore" has 120981 sinks.
[INFO CTS-0010]  Clock net "gated_clock_debug_clock_gate.out" has 1004 sinks.
[INFO CTS-0007] Net "serial_tl_0_clock" found for clock "serial_tl_0_clock".
[INFO CTS-0010]  Clock net "serial_tl_0_clock" has 91 sinks.
[INFO CTS-0008] TritonCTS found 3 clock nets.
[INFO CTS-0097] Characterization used 1 buffer(s) types.
[INFO CTS-0200] 11 placement blockages have been identified.
[INFO CTS-0027] Generating H-Tree topology for net clock_uncore.
[INFO CTS-0028]  Total number of sinks: 120981.
[INFO CTS-0090]  Sinks will be clustered based on buffer max cap.
[INFO CTS-0030]  Number of static layers: 0.
[INFO CTS-0020]  Wire segment unit: 1350  dbu (1 um).
[INFO CTS-0021]  Distance between buffers: 22 units (60 um).
[INFO CTS-0206] Best clustering solution was found from clustering size of 20 and clustering diameter of 50.
[INFO CTS-0019]  Total number of sinks after clustering: 7469.
[INFO CTS-0024]  Normalized sink region: [(3.44519, 5.22941), (737.393, 737.881)].
[INFO CTS-0025]     Width:  733.9481.
[INFO CTS-0026]     Height: 732.6516.
 Level 1
    Direction: Horizontal
    Sinks per sub-region: 3735
    Sub-region size: 366.9740 X 732.6516
[INFO CTS-0034]     Segment length (rounded): 184.
 Level 2
    Direction: Vertical
    Sinks per sub-region: 1868
    Sub-region size: 366.9740 X 366.3258
[INFO CTS-0034]     Segment length (rounded): 184.
 Level 3
    Direction: Horizontal
    Sinks per sub-region: 934
    Sub-region size: 183.4870 X 366.3258
[INFO CTS-0034]     Segment length (rounded): 92.
 Level 4
    Direction: Vertical
    Sinks per sub-region: 467
    Sub-region size: 183.4870 X 183.1629
[INFO CTS-0034]     Segment length (rounded): 92.
 Level 5
    Direction: Horizontal
    Sinks per sub-region: 234
    Sub-region size: 91.7435 X 183.1629
[INFO CTS-0034]     Segment length (rounded): 46.
 Level 6
    Direction: Vertical
    Sinks per sub-region: 117
    Sub-region size: 91.7435 X 91.5814
[INFO CTS-0034]     Segment length (rounded): 46.
 Level 7
    Direction: Horizontal
    Sinks per sub-region: 59
    Sub-region size: 45.8718 X 91.5814
[INFO CTS-0034]     Segment length (rounded): 22.
 Level 8
    Direction: Vertical
[... 136 lines cut ...]

Repair setup and hold violations...
TNS end percent 5
[INFO RSZ-0046] Found 650 endpoints with hold violations.
[INFO RSZ-0032] Inserted 940 hold buffers.
Placement Analysis
---------------------------------
total displacement        352.1 u
average displacement        0.0 u
max displacement           37.1 u
original HPWL        17820752.6 u
legalized HPWL       17821087.6 u
delta HPWL                    0 %

Warning: There are 12 input ports missing set_input_delay.
Warning: There are 8 output ports missing set_output_delay.
Warning: There are 323 unclocked register/latch pins.
Warning: There are 502 unconstrained endpoints.

==========================================================================
cts final report_design_area
--------------------------------------------------------------------------
Design area 401972 u^2 41% utilization.
Elapsed time: 1:20:19[h:]min:sec. CPU time: user 4814.64 sys 4.18 (99%). Peak memory: 10239348KB.
```

oharboe, 2024-03-06:
@maliberty Close? For my part the amount of output from CTS is reasonable. Output can always be improved everywhere, but no need for a github issue unless something is especially bad or there is a specific suggestion.

closed 2024-03-06 by maliberty