The-OpenROAD-Project/OpenROAD #4335: rtlmp should complete in a reasonable amount of time or fail quickly if is hopeless

oharboe, 2023-12-03:
### Description

Never completes...

```
./run-me-BoomTile-asap7-base.sh 
OpenROAD v2.0-11160-g042cdcfa4 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ORD-0030] Using 16 thread(s).
HierRTLMP Flow enabled...
Call Macro Placer  -halo_width 20 -report_directory bazel-bin/build//objects/asap7/BoomTile/base/rtlmp
Floorplan Outline: (0.0, 0.0) (4740.732, 4740.732),  Core Outline: (2.052, 2.16) (4738.716, 4738.5)
Traversed logical hierarchy
	Number of std cell instances: 2130872
	Area of std cell instances: 273473.66
	Number of macros: 7
	Area of macros: 4194455.50
	Area of macros with halos: 4534412.00
	Area of std cell instances + Area of macros: 4467929.00
	Core area: 22434450.00
	Design Utilization: 0.20
	Core Utilization: 0.01
	Manufacturing Grid: 1

[INFO MPL-0024] [Multilevel Autoclustering] Creating clustered netlist.
[INFO MPL-0039] [Coarse Shaping] Determining shape functions for clusters.
[INFO MPL-0028] [Hierarchical Macro Placement] Placing clusters and macros.
[no further output]
```

### Suggested Solution

Actionable error message if it is hopeless(what is wrong and what needs to be fixed) or complete in a practical amount of time. At most hours, not days.

AcKoucher, 2023-12-04:
@oharboe Write macro placement was already implemented in #4120. It was reported in #4068.

I'm updating Makefile options in ORFS PR #[1680](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1680). I'm including write_macro_placement there.

oharboe, 2023-12-04:
> @oharboe Write macro placement was already implemented in #4120. It was reported in #4068.
> 
> I'm updating Makefile options in ORFS PR #[1680](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1680) with -write_macro_placement included.

Ah yes, I recall now, thanks!

This issue is really about running times of rtlmp.

maliberty, 2023-12-07:
How are you running synthesis?  I'm surprised to see modules as small as
```
[DEBUG MPL-multilevel_autoclustering] Setting Cluster Metrics for core/ftq_arb: Num Macros: 0 Num Std Cells: 27
```

maliberty, 2023-12-07:
Are you setting
export SYNTH_HIERARCHICAL = 1
export RTLMP_FLOW = True
?

oharboe, 2023-12-07:
> Are you setting export SYNTH_HIERARCHICAL = 1 export RTLMP_FLOW = True ?

Yes, core/ftq_arb Verilog:

```verilog
module Arbiter_19(
  input        io_in_0_valid,
  input  [5:0] io_in_0_bits,
  input        io_in_1_valid,
  input  [5:0] io_in_1_bits,
               io_in_2_bits,
  output       io_in_2_ready,
  output [5:0] io_out_bits
);

  assign io_in_2_ready = ~(io_in_0_valid | io_in_1_valid);
  assign io_out_bits = io_in_0_valid ? io_in_0_bits : io_in_1_valid ? io_in_1_bits : io_in_2_bits;
endmodule
```

maliberty, 2023-12-07:
I see things have become inconsistent around MAX_UNGROUP_SIZE 
```
./designs/asap7/riscv32i/config.mk\013:export MAX_UNGROUP_SIZE ?= 1000
./designs/gf12/ariane/config.mk\05:export MAX_UNGROUP_SIZE ?= 10000
./designs/gf12/swerv_wrapper/config.mk\04:export MAX_UNGROUP_SIZE ?= 10000
./designs/gf12/tinyRocket/config.mk\06:export MAX_UNGROUP_SIZE ?= 1000
./designs/gf12/bp_quad/config.mk\07:export MAX_UNGROUP_SIZE ?= 1000
./designs/gf12/ariane133/config.mk\06:export MAX_UNGROUP_SIZE ?= 10000
./designs/nangate45/tinyRocket/config.mk\06:export MAX_UNGROUP_SIZE ?= 5000
./designs/tsmc65lp/tinyRocket/config.mk\06:export MAX_UNGROUP_SIZE ?= 5000
./designs/sky130hd_fakestack/ariane136/config.mk\06:export MAX_UNGROUP_SIZE = 10000
./platforms/nangate45/config.mk\025:export MAX_UNGROUP_SIZE ?= 10000
./scripts/synth_hier_report.tcl\031:  if { [info exist ::env(MAX_UNGROUP_SIZE)] && $::env(MAX_UNGROUP_SIZE) > 0 } {
./scripts/synth_hier_report.tcl\032:    set ungroup_threshold $::env(MAX_UNGROUP_SIZE)
./Makefile\0185:export MAX_UNGROUP_SIZE ?= 0
```
So in asap7 (and all non-ng45) it is defaulting to 0 which isn't great as every small module will be kept.   This will need some sorting out.

AcKoucher, 2023-12-08:
@oharboe FYI Even though #4360 solves the problem here, I think is important to comment that for this specific design we end up exhausting the mechanism that tries to estimate the best position for both macros and std cells in mpl2. As a result we use a more extreme mechanism to do so and this can deteriorate QoR.

Within this context, _I think #4354 can have a significant impact for us to get better results out of this type of situation._

PR #4360 (2023-12-12): mpl2: ensure we don't get stuck in cluster placement
Fix #4335 

We were getting stuck in when synchronizing the threads results.

Running Secure-CI
closed 2023-12-12 by maliberty