The-OpenROAD-Project/OpenROAD #4342: reduce number of rtlmp arguments

oharboe, 2023-12-05:
### Description

There are a lot of arguments to rtlmp, it would take forever to try them all.

### Suggested Solution

Make more of the arguments automatic and clearly mark debug options to rtlmp that the user shouldn't worry about.

oharboe, 2024-07-14:
@maliberty If we are agreed that this is a good idea and what we are moving towards, then I think this issue can be closed. 

This specific issue does not have any actionable suggestions towards that shared and understood goal in that case...

maliberty, 2024-07-15:
It is a good idea though I wouldn't say it is something that is getting a lot of cycles now.

closed 2024-07-15 by maliberty