The-OpenROAD-Project/OpenROAD #4347: Add `tclint` to OpenROAD CI

nmoroze, 2023-12-05:
### Description

This issue is a follow-up from discussion on #4337, and tracks adding [`tclint`](https://github.com/nmoroze/tclint.git) to the OpenROAD CI. 

`tclint` is a lint tool for Tcl that checks both style and functional errors. I've done some early experiments on this codebase, and with a baseline config there are ~23,000 lint violations across all the Tcl scripts. Nearly all of these are style-related. 

I think the best strategy would be to add linting incrementally to certain parts of the codebase. This will also give the maintainers a chance to review a reasonably-sized initial PR to make sure the style enforced by `tclint` is to their liking.

As an initial question, do any maintainers have a preference for a chunk of the Tcl codebase that might be a good candidate for the first linting pass? If given a pointer somewhere I'll open a PR that adds linting to CI for that part of the code and does the cleanup required to get it to pass.

maliberty, 2023-12-05:
How about src/rsz/src/Resizer.tcl as a sample

rovinski, 2023-12-05:
Should create a mirrored issue for ORFS.

On that note, it might be good to create separate issues for a Python linter as well.

@nmoroze are you planning on enhancing it reformat files similar to `clang-format -i`? That would be super useful.

nmoroze, 2023-12-06:
> Should create a mirrored issue for ORFS.

Done: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1686

> are you planning on enhancing it reformat files similar to clang-format -i? That would be super useful.

yeah, this was on my medium/long-term roadmap but I can consider prioritizing it, since it definitely seems useful for getting lint going on a large existing codebase like this one. At the very least, I can do some early experiments to see how hard it will be. I think `tclint` has most of the key pieces already, and "just" needs an emitter for the syntax tree to spit out well-formatted Tcl scripts. I tried to be mindful of this use case when designing the parse tree data structure to make sure it captures enough information, the devil's in the details and I have a feeling there will be some tricky bits :) 

I opened an issue to track this: https://github.com/nmoroze/tclint/issues/8

nmoroze, 2023-12-16:
I've submitted the first PR (#4397) towards this effort, cleaning up `Resizer.tcl` as suggested. 

One facilitating change to `tclint` was adding an option to not require an extra level of indentation within `namespace eval` blocks - this seems to be a common pattern in the OpenROAD code, and it seems to halve the overall number of lint violations found in the codebase.

If that PR looks good, I can go ahead and clean up the entire `src/rsz` directory unless anyone has alternate suggestions (which will hit a bunch of Tcl files in `src/rsz/test/`).  There are ~200 violations there, so it's chunkier but tractable.

maliberty, 2024-03-05:
Done

closed 2024-03-05 by maliberty