The-OpenROAD-Project/OpenROAD #4359: write_timing_model doesn't consider set_timing_derate

precisionmoon, 2023-12-08:
### Describe the bug

write_timing_model uses original timing from libraries without considering impact from set_timing_derate.

### Expected Behavior

It'd be good if write_timing_model and report_checks have the same behavior related to set_timing_derate.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.19.0-1027-gcp
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-11396-g137ea7c81
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
[... 154 lines cut ...]

# create SDC
create_clock -period 5 clk
set_input_delay 0.0 {e_in w_in} -clock clk
set_output_delay 0.0 {e_out w_out} -clock clk

puts "#### baseline ####"
report_checks
write_timing_model -library_name base_lib -cell base base.lib

set_timing_derate -late -cell_delay 0.5 [get_cells "inst_1"]
puts "#### after 0.5 derate ####"
report_checks
write_timing_model -library_name base0p5_lib -cell base0p5 base0p5.lib

set_timing_derate -late -cell_delay 2.0 [get_cells "inst_1"]
puts "#### after 2.0 derate ####"
report_checks
write_timing_model -library_name base2p0_lib -cell base2p0 base2p0.lib

### Relevant log output

```shell
base05p.lib and base2p0.lib don't reflect impact of set_timing_derate
```

precisionmoon, 2023-12-08:
[test.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/13610590/test.tar.gz)
estimate.tcl reproduces the issue

maliberty, 2023-12-08:
@tspyrou looks like one for Cherry

tspyrou, 2023-12-19:
I could ask Cherry to make it optional although I agree that normally the derating would be applied from the top.
@precisionmoon could you comment on the use case?

precisionmoon, 2023-12-19:
One simple use model is to add an option to write_timing_model like -use_derate to honor STD.  An info or warning message would be nice that explains that timing derates are being honored but this may lead to double-derating.

maliberty, 2025-08-12:
@tspyrou is there a conclusion on this?  close?

maliberty, 2026-03-24:
This issue has been automatically marked as stale because it has not had recent activity. It will be closed in 21 days if no further activity occurs. Remove the Stale label or comment to keep it open.

closed 2026-06-14 by github-actions[bot]