The-OpenROAD-Project/OpenROAD #4367: [WARNING ODB-0177] error: undefined via (via5_6_120_288_1_2_58_322) referenced

oharboe, 2023-12-12:
### Describe the bug

unzip https://drive.google.com/file/d/1puNriOUxW5JdMKytTCQN0I7J3lyoGvEI/view?usp=sharing

The error is output pretty much immediately(seconds):

```
$ ./run-me-BoomTile-asap7-base.sh 
OpenROAD v2.0-11423-g41f53d4db 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /home/oyvind/bazel_rules_hdl/megaboom/foo/platforms/asap7/lef/asap7_tech_1x_201209.lef
[INFO ODB-0223]     Created 24 technology layers
[INFO ODB-0224]     Created 9 technology vias
[INFO ODB-0226] Finished LEF file:  /home/oyvind/bazel_rules_hdl/megaboom/foo/platforms/asap7/lef/asap7_tech_1x_201209.lef
[INFO ODB-0222] Reading LEF file: /home/oyvind/bazel_rules_hdl/megaboom/foo/platforms/asap7/lef/asap7sc7p5t_28_R_1x_220121a.lef
[INFO ODB-0225]     Created 212 library cells
[INFO ODB-0226] Finished LEF file:  /home/oyvind/bazel_rules_hdl/megaboom/foo/platforms/asap7/lef/asap7sc7p5t_28_R_1x_220121a.lef
[INFO ODB-0222] Reading LEF file: bazel-bin/build/results/asap7/ebtb_128x40/base/ebtb_128x40.lef
[WARNING ODB-0177] error: undefined via (via5_6_120_288_1_2_58_322) referenced
[WARNING ODB-0177] error: undefined via (via5_6_120_288_1_2_58_322) referenced
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  bazel-bin/build/results/asap7/ebtb_128x40/base/ebtb_128x40.lef
[INFO ODB-0222] Reading LEF file: bazel-bin/build/results/asap7/array_256x64/base/array_256x64.lef
[WARNING ODB-0177] error: undefined via (via5_6_120_288_1_2_58_322) referenced
[WARNING ODB-0177] error: undefined via (via5_6_120_288_1_2_58_322) referenced
[INFO ODB-0225]     Created 1 library cells
[deleted]
```

### Expected Behavior

It should be either a warning or an error... Ideally the error should just be fixed...

Why does this happen and what should the user do about it? A more actionable error message would be great.

```
[WARNING ODB-0177] error: undefined via (via5_6_120_288_1_2_58_322) referenced
```

### Environment

```shell
OpenROAD v2.0-11423-g41f53d4db
```

### To Reproduce

See above

maliberty, 2023-12-12:
Are you sure the abstract you are using was generated after the fix to write the vias? This seems quite similar to what we had before.

oharboe, 2023-12-13:
> Are you sure the abstract you are using was generated after the fix to write the vias? This seems quite similar to what we had before.

@eder-matheus @maliberty Ah!

I see what I'm doing wrong.

I'm testing out OpenROAD w/Bazel on MegaBoom. I forgot to bump the ORFS_VERSION variable that I added. Whether or not I want to rebuild everything from scratch upon an ORFS/OpenROAD version bump is a decision that I want to make manually, otherwise it would always rebuild, effectively.

I'm going to do a version bump. If the error/warning is still there, I will file a fresh issue.

closed 2023-12-13 by oharboe