The-OpenROAD-Project/OpenROAD #4369: Remove bogus warning in floorplan_io.tcl about macro not being placed

oharboe, 2023-12-12:
### Description

I believe these warnings are bogus in 2_2_floorplan.log, there are no macros placed at this point.

Also, I thought that these warnings were removed?

```

OpenROAD v2.0-11117-g6061c4892 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING PPL-0015] Macro core/FpPipeline/fregfile/regfile_ext is not placed.
[WARNING PPL-0015] Macro core/iregfile/regfile_ext is not placed.
[WARNING PPL-0015] Macro dcache is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/bim/data_0_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/bim/data_1_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/bim/data_2_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/bim/data_3_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/btb/btb_0_0_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/btb/btb_0_1_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/btb/btb_0_2_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/btb/btb_0_3_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/btb/btb_1_0_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/btb/btb_1_1_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/btb/btb_1_2_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/btb/btb_1_3_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/btb/ebtb_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/btb/meta_0_0_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/btb/meta_0_1_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/btb/meta_0_2_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/btb/meta_0_3_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/btb/meta_1_0_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/btb/meta_1_1_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/btb/meta_1_2_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/btb/meta_1_3_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t/table_0_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t/table_1_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t/table_2_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t/table_3_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_1/table_0_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_1/table_1_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_1/table_2_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_1/table_3_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_2/table_0_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_2/table_1_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_2/table_2_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_2/table_3_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_3/table_0_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_3/table_1_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_3/table_2_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_3/table_3_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_4/table_0_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_4/table_1_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_4/table_2_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_4/table_3_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_5/table_0_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_5/table_1_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_5/table_2_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_0/tage/t_5/table_3_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_1/bim/data_0_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_1/bim/data_1_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_1/bim/data_2_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_1/bim/data_3_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_1/btb/btb_0_0_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_1/btb/btb_0_1_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_1/btb/btb_0_2_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_1/btb/btb_0_3_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_1/btb/btb_1_0_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_1/btb/btb_1_1_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_1/btb/btb_1_2_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_1/btb/btb_1_3_ext is not placed.
[WARNING PPL-0015] Macro frontend/bpd/banked_predictors_1/btb/ebtb_ext is not placed.
```

### Suggested Solution

Remove bogus warnings at this stage.

eder-matheus, 2023-12-12:
@oharboe This run uses a commit from November 16, and my update to remove these warnings is from November 23. The latest OR should behave as you expect.

oharboe, 2023-12-12:
@eder-matheus Thanks! I am working on multiple computers and got things mixed up.

closed 2023-12-12 by oharboe