The-OpenROAD-Project/OpenROAD #4373: write_lef crash

oharboe, 2023-12-13:
### Describe the bug

`tar -xzf` https://drive.google.com/file/d/1AfB6A1SsozJG7Tn6-nHftQgAqi8VhdIh/view?usp=sharing

```
$ openroad
OpenROAD v2.0-11428-g64228063b 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
openroad> read_db ./bazel-bin/build/results/asap7/BoomTile/base/deltaDebug_base_temp_3_place.odb
openroad> write_def a.def
openroad> write_lef b.lef
[INFO ORD-0034] More than one lib exists, multiple files will be written.
Signal 11 received
Stack trace:
 0# 0x0000558B27B2C483 in openroad
 1# 0x00007F500C842910 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbProperty::getPropOwner() in openroad
 3# odb::lefout::writePropertyDefinition(odb::dbProperty*) in openroad
 4# odb::lefout::writeObjectPropertyDefinitions(odb::dbObject*, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, short, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, short> > >&) in openroad
 5# odb::lefout::writePropertyDefinitions(odb::dbLib*) in openroad
 6# odb::lefout::writeHeader(odb::dbLib*) in openroad
 7# odb::lefout::writeTechAndLib(odb::dbLib*) in openroad
 8# ord::OpenRoad::writeLef(char const*) in openroad
 9# 0x0000558B27B8595E in openroad
10# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
11# 0x00007F5010F98C43 in /lib/x86_64-linux-gnu/libtcl8.6.so
12# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
13# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
14# ord::tclAppInit(Tcl_Interp*) in openroad
15# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
16# main in openroad
17# 0x00007F500C8280D0 in /lib/x86_64-linux-gnu/libc.so.6
18# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
19# _start in openroad
Segmentation fault (core dumped)
```

### Expected Behavior

no crash

### Environment

```shell
OpenROAD v2.0-11428-g64228063b
```

### To Reproduce

See above

maliberty, 2023-12-13:
This odb seems to be in a bad state.  It has a property "TYPE NWELL ;" but it seems to think it is attached to a dbScanChainObj which makes no sense.  How was this db generated?

oharboe, 2023-12-13:
> This odb seems to be in a bad state. It has a property "TYPE NWELL ;" but it seems to think it is attached to a dbScanChainObj which makes no sense. How was this db generated?

With deltaDebug as part of MegaBoom project. I wanted to run write_def/lef and read back in to see if that would reduce the .odb file size.

maliberty, 2023-12-13:
so the problem exists even before deltaDebug - how was the input odb generated?

maliberty, 2023-12-13:
Where you delta debugging this problem or another that then produced this problem on write_lef?

oharboe, 2023-12-13:
> so the problem exists even before deltaDebug - how was the input odb generated?

This is part of MegaBoom Bazel ORFS flow. It is the input, place, to the CTS crash as I recall.

maliberty, 2023-12-13:
Can you write_lef from the input db to delta debug?

oharboe, 2023-12-14:
@maliberty It seems to happen with just about any .odb file.

```
make DESIGN_CONFIG=designs/asap7/aes-block/config.mk floorplan
```

```
oyvind@gale:~/OpenROAD-flow-scripts/flow/results/asap7/aes-block/base$ openroad
OpenROAD v2.0-11428-g64228063b 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
openroad> read_db 2_1_floorplan.odb
openroad> write_lef foo.lef
[INFO ORD-0034] More than one lib exists, multiple files will be written.
Signal 11 received
Stack trace:
 0# 0x000055BEDD5F4483 in openroad
 1# 0x00007FB26F642910 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbProperty::getPropOwner() in openroad
 3# odb::lefout::writePropertyDefinition(odb::dbProperty*) in openroad
 4# odb::lefout::writeObjectPropertyDefinitions(odb::dbObject*, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, short, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, short> > >&) in openroad
 5# odb::lefout::writePropertyDefinitions(odb::dbLib*) in openroad
 6# odb::lefout::writeHeader(odb::dbLib*) in openroad
 7# odb::lefout::writeTechAndLib(odb::dbLib*) in openroad
 8# ord::OpenRoad::writeLef(char const*) in openroad
 9# 0x000055BEDD64D95E in openroad
10# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
11# 0x00007FB273D07C43 in /lib/x86_64-linux-gnu/libtcl8.6.so
12# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
13# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
14# ord::tclAppInit(Tcl_Interp*) in openroad
15# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
16# main in openroad
17# 0x00007FB26F6280D0 in /lib/x86_64-linux-gnu/libc.so.6
18# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
19# _start in openroad
```

PR #4382 (2023-12-14): odb: fix logic bug in dbTechVia::getViaParams
Fixes #4373
commit fee1af62 (2023-12-14): odb: fix logic bug in dbTechVia::getViaParams

Fixes #4373
closed 2023-12-14 by maliberty