The-OpenROAD-Project/OpenROAD #4383: Speed up CTS

oharboe, 2023-12-14:
### Description

unzip https://drive.google.com/file/d/1IKNjdXdUdjgcf1zjTXuyDntbyXLWgrO8/view?usp=sharing

```
./run-me-BoomTile-asap7-base.sh 
OpenROAD v2.0-11428-g64228063b 
[deleted]
[INFO CTS-0032]  Stop criterion found. Max number of sinks is 15.
[INFO CTS-0035]  Number of sinks covered: 4433.
[INFO CTS-0018]     Created 5337 clock buffers.
[INFO CTS-0012]     Minimum number of buffers in the clock path: 23.
[INFO CTS-0013]     Maximum number of buffers in the clock path: 24.
[INFO CTS-0014]     1 clock nets were removed/fixed.
[INFO CTS-0015]     Created 5336 clock nets.
[INFO CTS-0016]     Fanout distribution for the current clock = 0:1, 1:3, 2:10, 3:34, 4:50, 5:43, 6:55, 7:51, 8:61, 9:57, 10:56, 11:36, 12:32, 13:24, 14:18, 15:16, 16:11, 17:11, 18:7, 19:4, 20:4, 21:5, 22:3, 23:2, 24:3, 25:3, 26:2, 27:1, 29:1, 30:4338..
[INFO CTS-0017]     Max level of the clock tree: 9.
[INFO CTS-0098] Clock net "clock"
[INFO CTS-0099]  Sinks 131199
[INFO CTS-0100]  Leaf buffers 4430
[INFO CTS-0101]  Average sink wire length 2221.56 um
[INFO CTS-0102]  Path depth 23 - 24
[WARNING STA-0357] virtual clock clock_vir can not be propagated.
Warning: There are 540 unconstrained endpoints.
```

After ca. 1 minute, there is no further output for quite some time...

oharboe, 2023-12-14:
> How is this different from #4333 ?

One is a feature request for progress info, the other to speed up CTS.

maliberty, 2023-12-18:
@precisionmoon would you start by looking at the skew show in 
[image]
(from https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/discussions/1710) before we start on runtime.

maliberty, 2023-12-18:
@precisionmoon see my further comments in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/discussions/1710

maliberty, 2023-12-18:
with the current settings there are many buffers on the macros:
[image]

but if I add -obstruction_aware I see:
[image]

Some objects are now outside the die area which is wrong.  Some are still on top of macros too.

precisionmoon, 2023-12-18:
Ok. Let me take a look.

precisionmoon, 2024-01-02:
[PR 4383](https://github.com/The-OpenROAD-Project/OpenROAD/pull/4459) should fix the issue of cells getting placed outside core area and cells on top of hardmacros.

maliberty, 2024-01-02:
@precisionmoon with #4459 and -obstruction_aware  does the skew improve?

precisionmoon, 2024-01-02:
> @precisionmoon with #4459 and -obstruction_aware does the skew improve?

Worst skew remains at 58.56 immediately after CTS with and without -obstruction_aware, even after #4459.
Clock clock
Latency      CRPR       Skew
core/fp_rename_stage/_7310_/CLK ^
 451.28
core/fp_rename_stage/freelist/_45073_/CLK ^
 392.72      0.00      58.56