The-OpenROAD-Project/OpenROAD #4391: global routing congestion.rpt has DRC markers in a strange place

oharboe, 2023-12-15:
### Describe the bug

Why are the DRC markers (white x's) at the top and not in the middle of the congested area?

[image]

### Expected Behavior

Congestion markers in the middle of the congested area, not at the edge of the design.

### Environment

```shell
OpenROAD v2.0-11454-g34687e2a9
```

### To Reproduce

See above

maliberty, 2023-12-15:
Would you zoom in on the area with the markers and post the image.  When zoomed way out we use a X to make things visible.  There could be congestion near the edge of the block.

oharboe, 2023-12-15:
> Would you zoom in on the area with the markers and post the image. When zoomed way out we use a X to make things visible. There could be congestion near the edge of the block.

I no longer have this available on my computer, but using the reproduction case here you can. I didn't see anything there though, the area looked empty. There were square DRC congestion boxes at the top edge.

eder-matheus, 2024-01-06:
@oharboe I've run GRT with the files you've attached, and the congestion markers generated in this run are not the same as the one you attached:
[image]

Is it possible that you loaded a different congestion report file by mistake?

oharboe, 2024-01-07:
entirely...

closed 2024-01-07 by oharboe