The-OpenROAD-Project/OpenROAD #4394: Convert more mpl2 messages to debug output

oharboe, 2023-12-15:
### Description

Debug output...

```
[INFO MPL-0024] [Multilevel Autoclustering] Creating clustered netlist.
[INFO MPL-0039] [Coarse Shaping] Determining shape functions for clusters.
[INFO MPL-0028] [Hierarchical Macro Placement] Placing clusters and macros.
```

### Suggested Solution

Convert to debug output...

AcKoucher, 2023-12-15:
@oharboe My intention with these specific messages were just to inform the user of each stage of HierRTLMP flow. Of course there's no problem in removing them, but is it really the case?

oharboe, 2023-12-15:
> @oharboe My intention with these specific messages was just to inform the user of each stage of HierRTLMP flow. Of course there's no problem in removing them, but is it really the case?
> 
> Clarifying: It was just so that the user had some sort of idea of what MPL is doing. Again, if desired, I'll remove them.

Progress messages that give some real indication of time remaining are good, but otherwise removing noise from output helps make useful actionable.information more availble.

maliberty, 2023-12-16:
Its not always clear how long something will take so displaying the steps can be a reasonable proxy if it isn't too noisy.

oharboe, 2023-12-16:
> Its not always clear how long something will take so displaying the steps can be a reasonable proxy if it isn't too noisy.

These messages arent particularly helpful in this regard.... The first are printed, then ca 1hr passes (megaboom) and the rest are printed.

PR #4410 (2023-12-20): mpl2: refactoring of the top level function & removing more messages
Resolve #4394

My intention with the names of each stage method (e. g., runCoarseShaping) is to make it easier if we want split the HierRTLMP class into different classes for each engine at some point.

With this new top level interface it became easy to transform the root into a HardMacroCluster and get rid of that weird check for the macro only case, so I also did that.
closed 2023-12-20 by maliberty