The-OpenROAD-Project/OpenROAD #4395: [WARNING PDN-0110] No via inserted between M5 and M6 at (543.4920, 828.6400) - (543.6120, 828.6720) on VSS

oharboe, 2023-12-15:
### Describe the bug

untar https://drive.google.com/file/d/1v49C3TCcUf6ryJWWb6q8xnZ5cynolUdo/view?usp=drive_link

```
$ run-me-BoomTile-asap7-base.sh
OpenROAD v2.0-11454-g34687e2a9 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO PDN-0001] Inserting grid: top
[WARNING PDN-0110] No via inserted between M5 and M6 at (543.4920, 828.6400) - (543.6120, 828.6720) on VSS
[WARNING PDN-0110] No via inserted between M5 and M6 at (1251.9720, 1075.1280) - (1252.0920, 1075.1680) on VSS
[WARNING PDN-0110] No via inserted between M5 and M6 at (1329.7320, 1075.1280) - (1329.7340, 1075.1680) on VSS
```

### Expected Behavior

No warning

### Environment

```shell
OpenROAD v2.0-11454-g34687e2a9
```

### To Reproduce

See above

maliberty, 2023-12-16:
@gadfort do you have time to look at this?

maliberty, 2023-12-23:
These all look similar to 
[image]

The straps are too close to the macro to add a via and should have been pruned back.  It won't cause you any problem but we should avoid the issue for cosmetics.

gadfort, 2024-01-23:
@oharboe you might want to increase the halo for the macro to ensure PDN does not attempt to insert that set of straps.
I'm not able to run this at the moment due to the size of it.

oharboe, 2024-01-23:
@gadfort So if I understand corr

oharboe, 2024-01-23:
@gadfort So if I understand correctly, everything is working, but the warning does not offer your advice above. Should it?

closed 2024-01-23 by oharboe

gadfort, 2024-01-23:
The failed via report should contain some information, but it's possible that could be better documented.