The-OpenROAD-Project/OpenROAD #4402: PSM power grid check lists false-positive disconnected nodes inside macros

donn, 2023-12-18:
### Describe the bug

Trying a smoke-test multi-macro design with core rings: `check_power_grid -net VPWR` reports a multitude of disconnected PDN nets. Key is, the nets are not in fact disconnected there are vias at the claimed location: see this example

[image]

When disabling the check, the design does indeed pass LVS.

### Expected Behavior

A design passing LVS should not fail the PSM power grid check.

### Environment

```shell
Git commit: 75f2f325b7a42e56a92404f33af8e96530d9b202
kernel: Darwin 23.1.0
os: macOS 14.1.1
cmake version 3.27.7
CMake Deprecation Warning at third-party/abc/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Warning:
  Manually-specified variables were not used by the project:

    VERBOSE

-- The CXX compiler identification is Clang 16.0.6
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/zh206nri10wwp68g7qf9ccya1irynx5g-clang-wrapper-16.0.6/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-11533-g75f2f325b
-- System name: Darwin
-- Compiler: Clang 16.0.6
-- Build type: RELEASE
-- Install prefix: /var/empty/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is Clang 16.0.6
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/zh206nri10wwp68g7qf9ccya1irynx5g-clang-wrapper-16.0.6/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
[... 87 lines cut ...]
-- TCL header: /nix/store/l1rhkhykl3vzqipkk1c7wm6qanswxnrk-tcl-8.6.13/include/tcl.h
-- Found Eigen3: /nix/store/wx9kzvpd6ck59i9n2rg2jgir3km31n8b-eigen-3.4.0/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- GUI is enabled
-- Charts widget is not enabled
-- Found Boost: /nix/store/vf1zzbfy97i3awvva9yb2559vii2vc47-boost-1.83.0-dev/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found version "1.83.0") found components: serialization 
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /nix/store/vf1zzbfy97i3awvva9yb2559vii2vc47-boost-1.83.0-dev/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.78")  
-- TCL library: /nix/store/l1rhkhykl3vzqipkk1c7wm6qanswxnrk-tcl-8.6.13/lib/libtcl.dylib
-- TCL header: /nix/store/l1rhkhykl3vzqipkk1c7wm6qanswxnrk-tcl-8.6.13/include/tcl.h
-- Found Boost: /nix/store/vf1zzbfy97i3awvva9yb2559vii2vc47-boost-1.83.0-dev/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found version "1.83.0") found components: serialization system thread 
-- Removing MPL2 and PAR to avoid run time fatal error.
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Removing PAR_PY to avoid run time fatal error.
-- Configuring done (12.0s)
-- Generating done (0.7s)
-- Build files have been written to: /tmp/tmp.iWDflSEtcm
```

### To Reproduce

You may use the very short `./run.sh` or load the odb into the GUI:

[repro.tgz](https://github.com/The-OpenROAD-Project/OpenROAD/files/13705445/repro.tgz)

maliberty, 2023-12-18:
I believe LVS passes but the design is oddly constructed.  The instance has pins:
[image]
while at the top level you have:
[image]
They shapes are connected only through the instance pin shapes.   Is this intended?  How was this grid constructed?

donn, 2023-12-24:
@maliberty Apologies for the late response. Here's the reproducible for the top-level PDN generation: 

[repro.tgz](https://github.com/The-OpenROAD-Project/OpenROAD/files/13761392/repro.tgz)

Fully expecting something here to not make sense given your confusion 😅

maliberty, 2024-01-08:
any response to "Is this intended? How was this grid constructed?"

maliberty, 2024-01-08:
nvm

maliberty, 2024-01-08:
@gadfort any thoughts on this?  It appears to be trying to connect the shapes that in the ring pin of the instance.  Oddly it doesn't happen in both rings.

maliberty, 2024-01-08:
@gadfort your shape merging idea would fix the psm issue here.

gadfort, 2024-01-08:
@maliberty Ill take a look tomorrow if I can

PR #4850 (2024-04-09): psm: rework solver
Changes:
- reworks the IR solver to speed it up on larger designs
- adds caching of networks to speed up analysis across corners
- add alternative voltage source types (bumps, straps, and full for now)
- heatmap can select different nets/corners/layers

Breaking changes:
- The previous implementation would "snap" instances to the power grid, this does not and therefore `check_power_grid` will return an error if the places instances are not detailed placed.

Metrics:
- ariane136 (previous implementation takes about ~360s, this PR takes ~50s.
- on small designs the speedup wil [... cut]
closed 2024-04-09 by maliberty