The-OpenROAD-Project/OpenROAD #4406: Reduce CTS skew

oharboe, 2023-12-18:
### Description

[image]

To reproduce, unzip https://drive.google.com/file/d/1jSxumDJ_oWN5MEXX24faCkrg9UB5qNnc/view?usp=sharing, then run (ca. 4000s):

```
./run-me-BoomTile-asap7-base.sh
```

### Suggested Solution

Improve skew in CTS with default/automatic arguments.

maliberty, 2024-03-05:
@precisionmoon this is somewhat old - not sure if your recent work has solved this.

precisionmoon, 2024-03-06:
Let me re-run this with the latest version of CTS.

precisionmoon, 2024-03-06:
Here are some observations:
1) There are no macro sinks here, only registers.   So -insertion_delay doesn't help.
2) Skew gets worse after timing repair.
3) clustering diameter and size are no longer required.   Removing these options to enable auto clustering slightly improved skew.

This is clock tree before timing repair.
[image]

This is clock tree after timing repair.
[image]

There is no short-term quick solution.  We need to have a more holistic solution where timing repair can be done within CTS such that optimization is clock-tree aware.

Thanks.

oharboe, 2024-03-06:
@precisionmoon When you say holistic solution, does that include taking advantage of "useful skew" #4656 ?

If so, then I propose closing this issue as it covered by #4656. This is perhaps a useful testcase, but it is not a separate issue, as such.

precisionmoon, 2024-03-06:
We will keep this issue open for now.   Useful skew is a different solution.

maliberty, 2024-03-06:
There is a huge change in insertion delay before and after timing repair.  Do you know what accounts for that?  Is it related to repair_clock_net?

precisionmoon, 2024-03-06:
I didn't see any macro sinks, so where did you see insertion delay changes?   There are some "wire" buffers on clock trees that degrade clock skew.

maliberty, 2024-03-06:
I meant that the scale on the left side of the graphs is substantially different ~400 vs ~1400

precisionmoon, 2024-03-06:
Yes, the latencies increased after repair_clock_net.