The-OpenROAD-Project/OpenROAD #4407: Speed up global route

oharboe, 2023-12-18:
### Description

Global route takes at least 2x longer than detailed route here.

To reproduce(ca. 8000s in global route, vs. 4000s in detailed route):

unzip https://drive.google.com/file/d/1WmmWfZ_Ctb4wOl-bzoMhFBcu6pD-j0iN/view?usp=sharing, run:

```
./run-me-BoomTile-asap7-base.sh
```

### Suggested Solution

Speed up global route.

eder-matheus, 2023-12-19:
The biggest problem here is the incremental grt. It can be very long since multiple incremental runs of the GRT are made to repair timing, setup, hold violations, etc.

We'll try running a profiler to identify the bottlenecks and how we can improve them.

oharboe, 2023-12-19:
> The biggest problem here is the incremental grt. It can be very long since multiple incremental runs of the GRT are made to repair timing, setup, hold violations, etc.
> 
> We'll try running a profiler to identify the bottlenecks and how we can improve them.

O(n^2)?

eder-matheus, 2023-12-19:
> O(n^2)?

The major problem is that it may take thousands of iterations to complete the incremental groute. If you want to test, you can disable the incremental groute by adding the following to the config.mk file of your design:
`export SKIP_INCREMENTAL_REPAIR = 1`

But in any case, we'll still look at how to make it faster.

oharboe, 2023-12-20:
Example of where global route is stuck after 8 hours:

```
[INFO GRT-0018] Total wirelength: 26097568 um
[INFO GRT-0014] Routed nets: 1213301
[WARNING STA-0357] virtual clock clock_vir can not be propagated.
Warning: There are 135660 unclocked register/latch pins.
Warning: There are 135902 unconstrained endpoints.

==========================================================================
global route pre repair design report_design_area
--------------------------------------------------------------------------
Design area 1364898 u^2 51% utilization.
Perform buffer insertion...
[INFO RSZ-0058] Using max wire length 162um.
[no further output]
```

[image]

eder-matheus, 2024-03-15:
@oharboe After merging the PR https://github.com/The-OpenROAD-Project/OpenROAD/pull/4728, we did notice a good improvement in the runtime of incremental grt in our designs. Could you try it on your end?

oharboe, 2024-03-15:
@eder-matheus If you think this test case is exhausted, please close the issue as fixed.

I will over the next months be doing many global routes, I suppose and I will file a new github issue if I run into something that is surprisingly slow.

eder-matheus, 2024-03-15:
> @eder-matheus If you think this test case is exhausted, please close the issue as fixed.
> 
> I will over the next months be doing many global routes, I suppose and I will file a new github issue if I run into something that is surprisingly slow.

Sounds good! We'll also keep looking into more hotspots of the incremental grt. Hopefully, you will not face too many issues regarding runtime :)

closed 2024-03-15 by eder-matheus