The-OpenROAD-Project/OpenROAD #4425: report_clock_min_period: clock period_min = 0.00 fmax = inf

oharboe, 2023-12-21:
### Describe the bug

untar 

```
$ make ... open_2_1_floorplan.odb
[deleted]
OpenROAD v2.0-11553-gec214a37b 
[deleted]
openroad> report_clock_min_period
clock period_min = 0.00 fmax = inf
clock_vir period_min = 0.00 fmax = inf
openroad> 
```

### Expected Behavior

minimum clock period should not be 0.

### Environment

```shell
OpenROAD v2.0-11553-gec214a37b
```

### To Reproduce

See above

maliberty, 2023-12-23:
This looks bad:
```
Warning: There are 135660 unclocked register/latch pins.
Warning: There are 135902 unconstrained endpoints.
```

maliberty, 2023-12-23:
Is this right
```
[WARNING ORD-2011] LEF master cc_banks_8192x64 has no liberty cell.
[WARNING ORD-2011] LEF master cc_dir_1024x16 has no liberty cell.
[WARNING ORD-2011] LEF master data_data_40x128 has no liberty cell.
[WARNING ORD-2011] LEF master regfile_128x65 has no liberty cell.
[WARNING ORD-2011] LEF master regfile_128x64 has no liberty cell.
[WARNING ORD-2011] LEF master BoomNonBlockingDCache has no liberty cell.
[WARNING ORD-2011] LEF master data_2048x2 has no liberty cell.
[WARNING ORD-2011] LEF master btb_128x14 has no liberty cell.
[WARNING ORD-2011] LEF master ebtb_128x40 has no liberty cell.
[WARNING ORD-2011] LEF master meta_128x30 has no liberty cell.
[WARNING ORD-2011] LEF master table_128x11 has no liberty cell.
[WARNING ORD-2011] LEF master table_256x12 has no liberty cell.
[WARNING ORD-2011] LEF master table_128x13 has no liberty cell.
[WARNING ORD-2011] LEF master ghist_40x64 has no liberty cell.
[WARNING ORD-2011] LEF master meta_40x120 has no liberty cell.
[WARNING ORD-2011] LEF master ICache has no liberty cell.
[WARNING ORD-2011] LEF master l2_tlb_ram_0_512x45 has no liberty cell.
[WARNING ORD-2011] LEF master ram_256x8 has no liberty cell.
```

maliberty, 2023-12-23:
clock is only connected to 4 FF:
[image]

I think you just don't have any constrained paths to generate a min clock period from.

oharboe, 2023-12-23:
Thanks!

There are three top level clocks. I need to update my .sdc file.

closed 2023-12-23 by oharboe