The-OpenROAD-Project/OpenROAD #4426: Failed to fix all DRC errors in detailed routing

oharboe, 2023-12-21:
### Describe the bug

This is from https://github.com/The-OpenROAD-Project/megaboom/commit/bd036b21c9217355c31a34a2a2ef41c049586a91

untar https://drive.google.com/file/d/1RQeOoC-Al2PHKsCDgaIlKjwMAQz-NTXK/view?usp=sharing

load into GUI and load DRC report

[image]

[image]

Route guides for net:

[image]

### Expected Behavior

No DRC errors

### Environment

```shell
OpenROAD v2.0-11553-gec214a37b
```

### To Reproduce

See above

maliberty, 2023-12-21:
The guide looks to be very close to a macro where you can't legally drop a via.  @eder-matheus any thoughts on the guide?

eder-matheus, 2023-12-22:
> The guide looks to be very close to a macro where you can't legally drop a via. @eder-matheus any thoughts on the guide?

Yeah, they look too close to the macro. I'll try the macro extension option to see if it resolves and will investigate how to improve our resource modeling to avoid such guides so near the macros.

eder-matheus, 2024-01-02:
@oharboe Can you add a reproducible of this issue? Having only the final ODB for the routing will not be enough for debugging the global routing of this design.

oharboe, 2024-03-20:
@eder-matheus I think you fixed this?

closed 2024-03-20 by oharboe