The-OpenROAD-Project/OpenROAD #4433: Add Charts even if qt charts is not installed, but with a text saying qt charts not installed

oharboe, 2023-12-22:
### Description

If there are features that are invisible, because some dependency was not installed at compile time, how can the user know about them?

### Suggested Solution

Add Charts even if qt charts is not installed, but with a text saying qt charts not installed. This will tell users what features they are missing out on and think to look up the documentation to build OpenROAD correctly

maliberty, 2023-12-22:
cmake does print a message but this is better marketing.

rovinski, 2023-12-23:
Vim has a feature when running `:version` that displays all of the options it was compiled with. Perhaps something similar could be done with OR? Like `openroad --version` prints out

```
OpenROAD v2.0-11454-g34687e2a9
Features included (+) or not (-):
-charts -eqy +GPU +GUI +Python  ...
```

closed 2024-01-15 by eder-matheus

maliberty, 2024-01-15:
> Vim has a feature when running `:version` that displays all of the options it was compiled with. Perhaps something similar could be done with OR? Like `openroad --version` prints out
> 
> ```
> OpenROAD v2.0-11454-g34687e2a9
> Features included (+) or not (-):
> -charts -eqy +GPU +GUI +Python  ...
> ```

This is a nice idea for text mode but should be a separate issue.

PR #4535 (2024-01-15): gui: add charts event when qtcharts is not installed
Resolves #4433 

With these changes, Charts will always be built within the GUI, but if QtCharts is not installed, a message will be displayed "QtCharts is not installed".

[image]