The-OpenROAD-Project/OpenROAD #4434: Build and setup instructions should build "eqy"

oharboe, 2023-12-22:
### Description

Follow: https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildLocally.html

```
Cloning into 'eqy'...
remote: Enumerating objects: 1729, done.
remote: Counting objects: 100% (192/192), done.
remote: Compressing objects: 100% (145/145), done.
remote: Total 1729 (delta 68), reused 81 (delta 44), pack-reused 1537
Receiving objects: 100% (1729/1729), 1.05 MiB | 6.77 MiB/s, done.
Resolving deltas: 100% (1105/1105), done.
Note: switching to '0cc2ff0'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 0cc2ff0 Merge pull request #46 from YosysHQ/jix/dependent-shell-pattern-escape
yosys-config --build src/eqy_combine.so  src/eqy_combine.cc 
make: yosys-config: No such file or directory
make: *** [Makefile:26: src/eqy_combine.so] Error 127
```

Run:

```
make DESIGN_CONFIG=designs/asap7/aes/config.mk route
```

I get:

```
[deleted]
[INFO RSZ-0045] Inserted 1093 buffers, 1 to split loads.
[INFO RSZ-0041] Resized 275 instances.
[INFO RSZ-0043] Swapped pins on 416 instances.
[INFO RSZ-0049] Cloned 174 instances.
[WARNING RSZ-0062] Unable to repair all setup violations.
[INFO RSZ-0033] No hold violations found.
Error: cts.tcl, 111 couldn't execute "eqy": no such file or directory
Command exited with non-zero status 1
Elapsed time: 1:16.98[h:]min:sec. CPU time: user 76.77 sys 0.19 (99%). Peak memory: 277960KB.
make[1]: *** [Makefile:705: do-4_1_cts] Error 1
make: *** [Makefile:705: results/asap7/aes/base/4_1_cts.odb] Error 2
```

### Suggested Solution

Should work, dependencies should be installed

maliberty, 2023-12-22:
I thought the dependency installer does this now @vvbandeira

oharboe, 2023-12-22:
> I thought the dependency installer does this now @vvbandeira

Yes, but it fails...

princeofgambit, 2024-01-07:
[INFO RSZ-0049] Cloned 2 instances.
[WARNING RSZ-0062] Unable to repair all setup violations.
[INFO RSZ-0033] No hold violations found.
Error: cts.tcl, 111 couldn't execute "eqy": no such file or directory
Command exited with non-zero status 1
Elapsed time: 0:04.16[h:]min:sec. CPU time: user 4.04 sys 0.07 (98%). Peak memory: 105980KB.
make[1]: *** [Makefile:716: do-4_1_cts] Error 1
make: *** [Makefile:716: results/nangate45/gcd/base/4_1_cts.odb] Error 2
user@ubuntu:~/Desktop/OpenROAD-flow-scripts/flow$ make gui_final
make: *** No rule to make target 'gui_6_final.odb', needed by 'gui_final'.  Stop.
user@ubuntu:~/Desktop/OpenROAD-flow-scripts/flow$ 

same failure...Why?I am new to openroad

vvbandeira, 2024-01-16:
@oharboe and @princeofgambit 
Could you try again with the latest code? Please also make sure to `source env.sh` before running the flow if you used `sudo ./setup.sh` to install the dependencies.

pandey-shruti, 2024-03-01:
Hi @maliberty and @vvbandeira,

I am also getting the same error.  I am using the latest code and also ran source env.sh. Please look below for details:
Command 1: make DESIGN_CONFIG=./designs/nangate45/aes/config.mk

Error: cts.tcl, 111 couldn't execute "eqy": no such file or directory
Command exited with non-zero status 1
Elapsed time: 2:08.71[h:]min:sec. CPU time: user 128.63 sys 0.07 (99%). Peak memory: 189940KB.
Makefile:716: recipe for target 'do-4_1_cts' failed
make[1]: *** [do-4_1_cts] Error 1
Makefile:716: recipe for target 'results/nangate45/aes/base/4_1_cts.odb' failed
make: *** [results/nangate45/aes/base/4_1_cts.odb] Error 2
Please help me with to resolve the error. 

Thanks,
Shruti

vijayank88, 2024-03-01:
@pandey-shruti 
run `sudo ./setup.sh`, that way `dependencies` folder will be created and installed yosys and eqy.

stefanottili, 2024-03-01:
This doesn't work on MacOS M1.
Running setup.sh didn't install eqy.
It installs tools/LSOracle and tools/yosys and then the OR build is broken.

I had to fix tools/OpenROAD/etc/DependencyInstaller.sh to get ortools 9.8 on macOS Sonoma, there is a pending bug report about this that didn't make it yet.

+    orToolsVersionBig=9.8
+    orToolsVersionSmall=${orToolsVersionBig}.3296
+        _installOrTools "macOS" "14.1" $(uname -m)
         
I did blow away my previously working tools/OpenROAD/build directory and used ./build_openroad.sh --local --latest as usual.

The build then fails with something related to tcl and swig.
Anybody an idea what's causing this compile error ?

[ 86%] Building CXX object src/gui/CMakeFiles/gui.dir/src/tclCmdInputWidget.cpp.o
In file included from OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/tclCmdInputWidget.cpp:33:
In file included from OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/tclCmdInputWidget.h:50:
OpenROAD-flow-scripts/tools/OpenROAD/build/src/gui/tclSwig.h:1623:11: error: unknown type name 'Tcl_Size'; did you mean 'Tcl_Time'?
          Tcl_Size *vlptr = (Tcl_Size *) va_arg(ap, void *);
          ^~~~~~~~
          Tcl_Time

maliberty, 2024-03-06:
@oharboe are you still having this issue?

oharboe, 2024-03-06:
Yes:

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 23.10
Release:	23.10
Codename:	mantic
$ which yosys
/home/oyvind/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys
$ which -a yosys
/home/oyvind/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys
/usr/bin/yosys
/bin/yosys
$ sudo ./setup.sh
[deleted]
Lemon already installed.
spdlog already installed.
Cloning into 'eqy'...
remote: Enumerating objects: 1766, done.
remote: Counting objects: 100% (250/250), done.
remote: Compressing objects: 100% (173/173), done.
remote: Total 1766 (delta 101), reused 125 (delta 68), pack-reused 1516
Receiving objects: 100% (1766/1766), 1.01 MiB | 2.05 MiB/s, done.
Resolving deltas: 100% (1122/1122), done.
Note: switching to '8327ac7'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 8327ac7 Merge pull request #49 from YosysHQ/jix/submodule_example
yosys-config --build src/eqy_combine.so  src/eqy_combine.cc 
make: yosys-config: No such file or directory
make: *** [Makefile:26: src/eqy_combine.so] Error 127
```

PR #4744 (2024-03-07): etc: check for yosys-config
Fixes #4434
closed 2024-03-07 by vvbandeira