The-OpenROAD-Project/OpenROAD #4436: When CORE_UTILIZATION is used, use a minimum area such that io pins fit

oharboe, 2023-12-22:
### Description

Building on https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1722, it would be useful to use CORE_UTILIZATION, but that the area is increased so that the pins, respecting io constraints, can fit.

### Suggested Solution

CORE_UTILIZATION sets a minimum size, but the die area is increased such that io pins fit.

### Additional Context

...

oharboe, 2024-01-02:
@maliberty Perhaps the more idiomatic solution for openroad and ORFS is to print actionable error: increase CORE_UTILIZATION from xx to yy?

Settings specified should be used as is and the user should adjust if necessary based on information in errors, this reduces coupling between stages in the flow amongst other benefits.

maliberty, 2024-01-02:
It might be better to adjust CORE_MARGIN rather than spread things out more just to get more perimeter.

maliberty, 2024-01-02:
Though strictly speaking ppl doesn't know if you used CORE_UTILIZATION/CORE_MARGIN or DIE_AREA in ifp so we might need to suggest both.

oharboe, 2024-01-02:
> Though strictly speaking ppl doesn't know if you used CORE_UTILIZATION/CORE_MARGIN or DIE_AREA in ifp so we might need to suggest both.

Makes sense. An error message that it didnt fit would be an improvement even if much more specific advise in the error isnt approperiate.

eder-matheus, 2024-01-18:
@oharboe Maybe I'm missing something, but it´s unclear what the request is here. If I remember correctly, PPL already has error messages that suggest how much the edges should be increased.

Would the idea here be to suggest alternatives for CORE_UTILIZATION, CORE_MARGIN, and DIE_AREA when PPL can't fit the pins in the boundaries?

oharboe, 2024-01-18:
> @oharboe Maybe I'm missing something, but it´s unclear what the request is here. If I remember correctly, PPL already has error messages that suggest how much the edges should be increased.

If it shows a) how much it is today b) how much it would need to be as well as how many pins that fit and how many that were requested to fit, then the user can carry the water the rest of the way. There are going to be differences in the action to take depending on build system and how it is configured(ORFS, SiliconCompiler, etc.)

eder-matheus, 2024-01-18:
> If it shows a) how much it is today b) how much it would need to be as well as how many pins that fit and how many that were requested to fit, then the user can carry the water the rest of the way. [...]

Thanks for the clarification! I'll work on adding these pieces of information to PPL. I believe this will require updating a few error messages, like when constrained pins don't fit, when pin groups don't fit, and even when just regular pins will not fit.

oharboe, 2024-01-18:
Super! The general idea is that an error message should generally have all the information required to remedy the problem, i.e. the error should be actionable.

eder-matheus, 2024-01-19:
@oharboe I've started with the error where the die area will not fit all the pins. This is the new message:

`[ERROR PPL-0024] Number of IO pins (54) exceeds maximum number of available positions (52). Increase the die area to 12074.83 to have enough space for all the pins.`

Is it aligned with your request?

oharboe, 2024-01-19:
> @oharboe I've started with the error where the die area will not fit all the pins. This is the new message:
>
> `[ERROR PPL-0024] Number of IO pins (54) exceeds maximum number of available positions [...]

It should be to increase width/height because it may be desirable to change aspect and keep area for instance...

eder-matheus, 2024-01-19:
> It should be to increase width/height because it may be desirable to change aspect and keep area for instance...

So it would be more like what we have for the constraints:
`[WARNING PPL-0110] Constraint has 54 pins, but only 52 available slots. Increase the region in at least 0.52um.`

oharboe, 2024-01-19:
replace "the region" with "region on left/top/right/bottom".

the idea is to provide information that the user must anyway find out ...

eder-matheus, 2024-01-19:
> replace "the region" with "region on left/top/right/bottom".
> 
> the idea is to provide information that the user must anyway find out ...

This is not the final message. This is just an example from an existing message that I used to understand if the format is what you need.

oharboe, 2024-03-20:
@maliberty @eder-matheus Close? I get reasonable actionable error messages now with numbers on current and needed sizes.

closed 2024-03-20 by eder-matheus