The-OpenROAD-Project/OpenROAD #4441: Dumping estimated parasitics as SPEF

donn, 2023-12-25:
### Description

You can run `estimate_parasitics` prior to RCX extraction with OpenROAD, however, there isn't a way to dump this data to a SPEF file as far as we can tell. We're trying to bring up a PDK internally and it would be helpful for calibrating the RC values. 

### Suggested Solution

A way to dump the estimated parasitics would be ideal.

maliberty, 2023-12-26:
There isn't a method currently.  How would you use the spef?  If you just need something like total C that's simpler.

QuantamHD, 2023-12-28:
I'd also like this

donn, 2024-01-09:
> There isn't a method currently. How would you use the spef? If you just need something like total C that's simpler.

We're trying to compare the values pre and post-routing (and RCX) to try and calibrate the RC values to try to get the pre-routed values a bit closer to post-route extracted parasitics.

If total C (presuming you meant per-net) can also be dumped the same format pre and post routing and RCX, that would also be helpful as a starting point.

rovinski, 2024-01-09:
@donn take a look in ORFS where we have a script that does linear correlation between GPL, GRT, and RCX parasitics on both total cap and total res.

[`write_net_rc.tcl`](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/6b53853149d273909d7458f3f6dcba1397767454/flow/util/write_net_rc.tcl) contains a bunch of procs for extracting cap and res for each net.
[`write_net_rc_script.tcl`](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/6b53853149d273909d7458f3f6dcba1397767454/flow/util/write_net_rc_script.tcl) uses those procs to get the GPL, GRT, and RCX cap and res into CSV files.
[`correlateRC.py`](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/6b53853149d273909d7458f3f6dcba1397767454/flow/util/correlateRC.py) does a simple linear regression on the estimated cap and the extracted cap to generate a correlated `setRC.tcl`

[`make correlate_rc`](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/6b53853149d273909d7458f3f6dcba1397767454/flow/util/utils.mk#L46-L48) will run these scripts on a specific design
[`make correlate_rc_platform`](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/6b53853149d273909d7458f3f6dcba1397767454/flow/util/utils.mk#L53-L59) will run the correlation on all designs in an entire platform, rather than just on a single design. Although for some reason Make always wants to rebuild a design from scratch even if it exists.

maliberty, 2024-08-08:
@eder-matheus please help @joaomai with this as an initial project

closed 2024-09-11 by maliberty