The-OpenROAD-Project/OpenROAD #4449: Increase level of detail dynamically when the GUI is idle

oharboe, 2023-12-27:
### Description

When zooming in/out it is helpful with a low level of detail to speed up the GUI.

However, when the GUI is idle, it would be nice if the design rendered higher level of detail in the background.

Example:

1. unzip https://drive.google.com/file/d/1lcqYhsNwozFox4IzPegbkxAW9Kk4V4Lq/view?usp=sharing
2. See top level view, all black (good, because panning zooming at top level is fast):
[image]
3. zoom in, see more detail: 
[image]
4. It would be nice if, after a while, the top level rendered more detail.

### Suggested Solution

Quick zoom/pan hither and dither, then stop moving and wait for higher level of detail.

maliberty, 2023-12-27:
What defines "a low level of detail"?

oharboe, 2023-12-27:
> What defines "a low level of detail"?

Something that is rendered quickly enough to make the GUI interactive and intuitive. Some sort of heuristic is used today....

Ideally when panning around, rendering should be on the order of 500ms or less and then as user stops wiggling things around, things could take a long time(a few seconds to dozen of seconds w/heatmaps) to render in full detail.