The-OpenROAD-Project/OpenROAD #4457: "No paths found" when loading .odb and using "report_checks" while trying to generate an histogram

lustefan97, 2023-12-28:
### Describe the bug

I want to create a slack histogram using a previously placed and routed design. I have a script that read the liberty files for the PDK used in the design, extract RC parasitics and write a spef file, and report timing with `report_checks`, `report_wns`, etc. A warning `[WARNING GRT-0026] Missing route to pin <pin name>` is raised, and when trying the `report_checks`command is used a _No paths found_ message is displayed. Obviously trying to create an histogram based off of endpoints  result in a crash, but I know this is worked on

(Note that my IO pins are placed on the top metal layer, if that is of any help)

### Expected Behavior

Design loaded in, RC parasitic extracted and path found for timing analysis => creating a slack histogram

### Environment

```shell
Git commit: 31d7e3dc5ce306b899dd98172022fca8aebc024f
kernel: Linux 6.2.0-39-generic
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
```

### To Reproduce

[TestCase.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/13786918/TestCase.tar.gz)

Extract this file in _OpenROAD/test/_, enter the TestCase directory created, use `openroad -gui` and in the TCL command box, enter `source histogram_script.tcl`

maliberty, 2023-12-28:
You'll need to do read_spef as well to load the file.   Make sure you do " set_propagated_clock [all_clocks]" as well.

lustefan97, 2023-12-28:
I already use `read_spef` in the script that is provided in the test case. I did add `set_propagated_clock [all_clock]` to it, with no success. The "No path found" warning is still there, I did enter those two command manually to make sure that the file were loaded in, but still it resulted in a crash (but I am waiting to update with the new PR about it)

maliberty, 2023-12-28:
You haven't loaded any sdc so there are no timing constraints.  Also you have an odd character after 'else' that makes it invalid and so estimate_parasitics is running.  

Is there some reason you need to use your own script instead of ORFS?  Debugging your script is not a productive use of time.

closed 2023-12-28 by maliberty

lustefan97, 2023-12-28:
I am currently running OpenROAD on a server without any gui, yet I am interested in the histogram slack proposed by OpenROAD for timing analysis. This script was for displaying the timing histogram of a circuit based off of its .odb file acquired on the server. I am also currently still learning about EDAs and ASIC flow so I was more interested in tweaking a custom script rather than using a black box.

Thank you for your time, and sorry for the inconvenience