The-OpenROAD-Project/OpenROAD #4462: [ERROR DRT-0073] No access point for tile_prci_domain/tile_reset_domain/boom_tile/frontend/bpd/clock.

oharboe, 2023-12-29:
### Describe the bug

This is from https://github.com/The-OpenROAD-Project/megaboom/commit/f5dd1af7841cf1c811a51206bda6f36b446ff335

untar https://drive.google.com/file/d/1BVX_h7KegG6F6Qa48d2bkDXRW-nZoL5e/view?usp=sharing

```
$ ./run-me-DigitalTop-asap7-base.sh 
OpenROAD v2.0-11600-g1bb904876 
[deleted]
[INFO DRT-0077]   Complete 2000 pins.
[INFO DRT-0077]   Complete 3000 pins.
[ERROR DRT-0073] No access point for tile_prci_domain/tile_reset_domain/boom_tile/frontend/bpd/clock.
[ERROR DRT-0073] No access point for tile_prci_domain/tile_reset_domain/boom_tile/core/FpPipeline/clock.
[ERROR DRT-0073] No access point for tile_prci_domain/tile_reset_domain/boom_tile/frontend/icache/auto_master_out_a_bits_address[0].
[ERROR DRT-0073] No access point for tile_prci_domain/tile_reset_domain/boom_tile/dcache/auto_out_a_bits_address[0].
[ERROR DRT-0073] No access point for subsystem_l2_wrapper/l2/auto_ctrls_ctrl_in_a_bits_address[0].
Error: detail_route.tcl, 69 DRT-0073
openroad> 
```

### Expected Behavior

No error

### Environment

```shell
OpenROAD v2.0-11600-g1bb904876
```

### To Reproduce

See above

osamahammad21, 2023-12-31:
The pin tile_prci_domain/tile_reset_domain/boom_tile/frontend/bpd/clock is completely covered by routing obstruction on M4 and M5. So there is no way to route to the pin.
[image]
[image]

oharboe, 2024-01-03:
@osamahammad21 I think this might be a red herring. I'll refile a github issue with more detail if it recoccurs.

closed 2024-01-03 by oharboe