The-OpenROAD-Project/OpenROAD #4466: Infinite loop(?) in detailed routing

oharboe, 2023-12-31:
### Describe the bug

The detailed routing spends ca. 20 minutes to get through 0 and 1st iteration, then things proceed pretty quickly and it gets stuck(infinite loop?) in the 51st iteration.

To reproduce untar

```
$ NUM_CORES=16 ./run-me-FpPipeline-asap7-base.sh 
OpenROAD v2.0-11602-g18cbc8d2b 
[deleted]
[INFO DRT-0195] Start 1st optimization iteration.
    Completing 10% with 72941 violations.
    elapsed time = 00:01:43, memory = 23246.29 (MB).
    Completing 20% with 56039 violations.
    elapsed time = 00:03:00, memory = 23356.55 (MB).
    Completing 30% with 49203 violations.
    elapsed time = 00:03:48, memory = 23667.76 (MB).
    Completing 40% with 37493 violations.
    elapsed time = 00:05:51, memory = 24035.68 (MB).
    Completing 50% with 32494 violations.
    elapsed time = 00:06:17, memory = 22973.07 (MB).
    Completing 60% with 32494 violations.
    elapsed time = 00:07:57, memory = 23971.48 (MB).
    Completing 70% with 19858 violations.
    elapsed time = 00:09:16, memory = 23713.82 (MB).
    Completing 80% with 15360 violations.
    elapsed time = 00:10:07, memory = 24022.12 (MB).
    Completing 90% with 6137 violations.
    elapsed time = 00:12:09, memory = 24350.66 (MB).
    Completing 100% with 3126 violations.
    elapsed time = 00:12:31, memory = 23257.75 (MB).
[INFO DRT-0199]   Number of violations = 3592.
Viol/Layer          M1     M2     M3     V3     M4     V4     M5     V5
CutSpcTbl            0      0      0     55      0     11      0      2
EOL                  0    388      8      0      2      0      0      0
Metal Spacing       80     76    234      0      5      0     70      0
Recheck              6    259    183      0     18      0      0      0
Short                8    204      3      0      3      0     23      0
eolKeepOut           0   1902     38      0     11      0      3      0
[INFO DRT-0267] cpu time = 03:08:15, elapsed time = 00:12:35, memory = 22662.84 (MB), peak = 24556.07 (MB)
Total wire length = 1041540 um.
Total wire length on LAYER M1 = 0 um.
Total wire length on LAYER M2 = 296351 um.
Total wire length on LAYER M3 = 401157 um.
Total wire length on LAYER M4 = 197543 um.
Total wire length on LAYER M5 = 101650 um.
Total wire length on LAYER M6 = 30231 um.
Total wire length on LAYER M7 = 12775 um.
Total wire length on LAYER M8 = 1829 um.
Total wire length on LAYER M9 = 0 um.
Total wire length on LAYER Pad = 0 um.
Total number of vias = 1450472.
Up-via summary (total 1450472):.

------------------
 Active          0
     M1     469522
     M2     841059
     M3     114855
     M4      20500
     M5       3458
     M6        941
     M7        137
     M8          0
     M9          0
------------------
           1450472

[INFO DRT-0195] Start 2nd optimization iteration.
    Completing 10% with 3592 violations.
    elapsed time = 00:00:33, memory = 23634.98 (MB).
    Completing 20% with 3178 violations.
    elapsed time = 00:00:49, memory = 23311.73 (MB).
    Completing 30% with 3072 violations.
    elapsed time = 00:01:04, memory = 23409.81 (MB).
    Completing 40% with 2731 violations.
    elapsed time = 00:01:41, memory = 23313.86 (MB).
    Completing 50% with 2667 violations.
    elapsed time = 00:01:44, memory = 22406.85 (MB).
    Completing 60% with 2667 violations.
    elapsed time = 00:02:20, memory = 23354.41 (MB).
    Completing 70% with 2296 violations.
    elapsed time = 00:02:35, memory = 23064.76 (MB).
    Completing 80% with 2244 violations.
    elapsed time = 00:02:50, memory = 23353.98 (MB).
    Completing 90% with 1896 violations.
    elapsed time = 00:03:25, memory = 23494.27 (MB).
    Completing 100% with 1829 violations.
    elapsed time = 00:03:29, memory = 22406.82 (MB).
[INFO DRT-0199]   Number of violations = 2055.
Viol/Layer          M1     M2     V2     M3     V3     M4     V4     M5
CutSpcTbl            0      0      0      0     11      0      4      0
EOL                  0    239      0      5      0      0      0      0
Metal Spacing       45     46      0    111      0      0      0     70
Recheck              4    147      0    101      0      9      0      0
Short                1    106      1      3      0      1      0     23
eolKeepOut           0   1107      0     18      0      3      0      0
[INFO DRT-0267] cpu time = 00:51:43, elapsed time = 00:03:32, memory = 22087.69 (MB), peak = 24556.07 (MB)
[deleted]
[INFO DRT-0195] Start 50th optimization iteration.
    Completing 10% with 75 violations.
    elapsed time = 00:00:00, memory = 21224.53 (MB).
    Completing 20% with 51 violations.
    elapsed time = 00:00:08, memory = 21224.53 (MB).
    Completing 30% with 37 violations.
    elapsed time = 00:00:15, memory = 21224.53 (MB).
    Completing 40% with 37 violations.
    elapsed time = 00:00:15, memory = 21224.53 (MB).
    Completing 50% with 37 violations.
    elapsed time = 00:00:15, memory = 21224.53 (MB).
    Completing 60% with 37 violations.
    elapsed time = 00:00:15, memory = 21224.53 (MB).
    Completing 70% with 34 violations.
    elapsed time = 00:00:21, memory = 21224.53 (MB).
    Completing 80% with 24 violations.
    elapsed time = 00:00:24, memory = 21224.53 (MB).
    Completing 90% with 24 violations.
    elapsed time = 00:00:24, memory = 21224.53 (MB).
    Completing 100% with 24 violations.
    elapsed time = 00:00:24, memory = 21224.53 (MB).
[INFO DRT-0199]   Number of violations = 24.
Viol/Layer          M5
Metal Spacing       19
Short                5
[INFO DRT-0267] cpu time = 00:00:40, elapsed time = 00:00:24, memory = 21224.53 (MB), peak = 24556.07 (MB)
[deleted]
[INFO DRT-0195] Start 51st optimization iteration.
    Completing 10% with 24 violations.
    elapsed time = 00:00:00, memory = 21224.53 (MB).
[no further output]
```

### Expected Behavior

Complete in a reasonable amount of time

### Environment

```shell
OpenROAD v2.0-11602-g18cbc8d2b
```

### To Reproduce

See above

maliberty, 2023-12-31:
Did you observe a similar slow down in iter 44?  We periodically try a more aggressive rip up strategy.

maliberty, 2023-12-31:
Do you have a power rail too close to the pins?

maliberty, 2023-12-31:
There is no way this is routable:
[image]

maliberty, 2023-12-31:
Something is wrong as you have pdn stripes going outside the core area.  Please investigate earlier in the flow what is going wrong.

maliberty, 2023-12-31:
@osamahammad21 if we have drc between a pin shape and a fixed shape we should fail early.

osamahammad21, 2023-12-31:
> @osamahammad21 if we have drc between a pin shape and a fixed shape we should fail early.

Does that mean we need to check_drc before routing and check for such drcs?

maliberty, 2023-12-31:
Can we just check for such in iter0 in any markers generated?

osamahammad21, 2024-01-01:
That's a bit challenging as we normally skip possible violations among fixed shapes. And when we do route, we merge the pin shape with the net segment connecting to it under one gcNet (to have correct max rectangles and polygon edges).

oharboe, 2024-01-01:
> Something is wrong as you have pdn stripes going outside the core area. Please investigate earlier in the flow what is going wrong.

@maliberty To reproduce the problem of power strips being generated outside of the die area:

1. untar https://drive.google.com/file/d/1GKoMeZo_xHGD9DcGK5C1skHCowcvZgrc/view?usp=sharing
2. Run `./run-me-FpPipeline-asap7-base.sh`
3. Zoom in on top edge

[image]

[image]

maliberty, 2024-01-01:
I would be nice to stop a doomed run early.  I don't want to spend a lot of time looking for a rare case though.  If you see an easy/fast way to detect it then please do so otherwise c'est la vie.

oharboe, 2024-01-01:
> I would be nice to stop a doomed run early. I don't want to spend a lot of time looking for a rare case though. If you see an easy/fast way to detect it then please do so otherwise c'est la vie.

I can't recall detailed routing running into an infinite loop in this way, I'd be wary of adding conditional code for such a rare case... You'd definitely need automated test cases for such rare conditions or it will bit rot anyway.

oharboe, 2024-01-01:
@maliberty So the root cause is PDN, for which I created a reproduction case in https://github.com/The-OpenROAD-Project/OpenROAD/issues/4466#issuecomment-1873351593 ?

oharboe, 2024-01-01:
> Did you observe a similar slow down in iter 44? We periodically try a more aggressive rip up strategy.

Yes, I suppose... See bottom.

Lots of warnings at the top...

```
$ ./run-me-FpPipeline-asap7-base.sh 
OpenROAD v2.0-11602-g18cbc8d2b 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING STA-0357] virtual clock clock_vir can not be propagated.
Error: detail_route.tcl, 11 can't read "::env(NUM_CORES)": no such variable
openroad> 
oyvind@corona:~/megaboom/foo$ NUM_CORES=16 ./run-me-FpPipeline-asap7-base.sh 
OpenROAD v2.0-11602-g18cbc8d2b 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING STA-0357] virtual clock clock_vir can not be propagated.
[INFO ORD-0030] Using 16 thread(s).
detailed_route -output_drc bazel-bin//reports/asap7/FpPipeline/base/5_route_drc.rpt -output_maze bazel-bin//results/asap7/FpPipeline/base/maze.log -bottom_routing_layer M2 -top_routing_layer M9 -save_guide_updates -verbose 1 -drc_report_iter_step 5
[INFO DRT-0149] Reading tech and libs.
[WARNING DRT-0140] SpacingRange unsupported.
[WARNING DRT-0145] New SPACINGTABLE PARALLELRUNLENGTH overrides old SPACING rule.
[WARNING DRT-0145] New SPACINGTABLE PARALLELRUNLENGTH overrides old SPACING rule.
[WARNING DRT-0145] New SPACINGTABLE PARALLELRUNLENGTH overrides old SPACING rule.
[WARNING DRT-0145] New SPACINGTABLE PARALLELRUNLENGTH overrides old SPACING rule.
[WARNING DRT-0323] Via(s) in pin VDD of regfile_128x65 will be ignored
[WARNING DRT-0323] Via(s) in pin VSS of regfile_128x65 will be ignored

Units:                1000
Number of layers:     21
Number of macros:     213
Number of vias:       11
Number of viarulegen: 11

[INFO DRT-0150] Reading design.
Error: duplicated via definition for via5_6_120_288_1_2_58_322

Design:                   FpPipeline
Die area:                 ( 0 0 ) ( 349606 349606 )
Number of track patterns: 32
Number of DEF vias:       0
Number of components:     611538
Number of terminals:      1141
Number of snets:          2
Number of nets:           159000

[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[0]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[10]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[11]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[12]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[13]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[14]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[15]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[16]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[17]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[18]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[19]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[1]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[2]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[3]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[4]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[5]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[6]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[7]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[8]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_mispredict_mask[9]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_resolve_mask[0]
[WARNING DRT-0422] No routing tracks pass through the center of Term io_brupdate_b1_resolve_mask[10]
[... 3206 lines cut ...]
     M6        940
     M7        134
     M8          0
     M9          0
------------------
           1444731

[INFO DRT-0195] Start 44th optimization iteration.
    Completing 10% with 28 violations.
    elapsed time = 00:00:00, memory = 21128.20 (MB).
    Completing 20% with 28 violations.
    elapsed time = 00:05:30, memory = 21128.20 (MB).
    Completing 30% with 28 violations.
    elapsed time = 00:07:16, memory = 21128.20 (MB).
    Completing 40% with 28 violations.
    elapsed time = 00:07:16, memory = 21128.20 (MB).
    Completing 50% with 28 violations.
    elapsed time = 00:07:16, memory = 21128.20 (MB).
    Completing 60% with 28 violations.
    elapsed time = 00:07:16, memory = 21128.20 (MB).
    Completing 70% with 28 violations.
    elapsed time = 00:07:17, memory = 21128.20 (MB).
[did not observe the run any further]

```

maliberty, 2024-01-01:
```
add_pdn_ring   -grid {top} -layers {M5 M6} -widths {0.504 0.544} -spacings {0.096} -core_offset {0.504}
```

leads to rings outside the core area are there isn't enough room to put the rings between the core and die areas:
[image]

You'll either need to adjust the rings or the floorplan (or both).

oharboe, 2024-01-01:
> You'll either need to adjust the rings or the floorplan (or both).

Thanks!

Is there a reason there cant be an actionable error message here?

oharboe, 2024-01-02:
@maliberty With `CORE_UTILIZATION=30`, there is no way to adjust the amount of space between the core area and the die area, this is done automatically.

Shouldn't the spacing be adjusted to a legal amount automatically?

maliberty, 2024-01-02:
You set it through CORE_MARGIN

oharboe, 2024-01-02:
> You set it through CORE_MARGIN

Thanks!

Feature request: an actionable error message. "Error: No space for power ring, increase CORE_MARGIN"

(or the openroad speak equivalent from where the user should be able to figure out CORE_MARGIN)

oharboe, 2024-01-02:
@maliberty Tested some more.

Long story short: This is issue is now a feature request for adding an actionable error message when the PDN ring doesn't fit.

gadfort, 2024-01-03:
How come the pins were placed overlapping the power straps? That seems to be the actual issue here based on my read of this.

oharboe, 2024-01-03:
> How come the pins were placed overlapping the power straps? That seems to be the actual issue here based on my read of this.

Because CORE_MARGIN, CORE/DIE_AREA doesn't leave enough room for the power ring. The error message will tell you at the floorplanning stage so as to avoid having to wait all the way to routing before discovering this.

gadfort, 2024-01-03:
@oharboe that doesn't explain why PPL put in the pins on top of the straps

oharboe, 2024-01-03:
> @oharboe that doesn't explain why PPL put in the pins on top of the straps

What is PPL?

Pins are added before the PDN in ORFS.

gadfort, 2024-01-03:
If the pins are marked FIXED then PDN will avoid them, if they are just PLACED then PDN is expecting a second call to the pin placer tool (PPL) to place them properly.

oharboe, 2024-01-03:
> If the pins are marked FIXED then PDN will avoid them, if they are just PLACED then PDN is expecting a second call to the pin placer tool (PPL) to place them properly.

So... more things to investigate...

Still seems like a good idea to have an error message if the PDN ring can't be placed inside the die area...

gadfort, 2024-01-03:
@oharboe sure, I don't disagree (I just think the implementation in the PR is not quite consistent with the rest of the codebase), I generally don't like that PDN places shaped outside the DIE area, but OpenLANE relies on this behavior.

maliberty, 2024-01-03:
I'm guessing that ppl doesn't look for obstructions on the die edge as there generally aren't any (@eder-matheus is that right?).  Even if it did put the pins off the straps, they would not be on the edge of the block anymore which wouldn't be good.  You would have to access them from above which doesn't make sense.

I don't know about what OL is doing with such stripes.  Would you say more.

eder-matheus, 2024-01-03:
@maliberty PPL actually looks for obstructions, but I believe only placement obstructions. We iterate through them using this for loop:
`for (odb::dbObstruction* obstruction : getBlock()->getObstructions()) {`

I could extend this function to iterate through PDN shapes and ensure that PPL will not use tracks that overlap with these nets.

maliberty, 2024-01-03:
@eder-matheus those are routing obstructions rather than placement blockages.  I'm not sure this usage makes sense.  Let's see how the discussion concludes before doing any work.

eder-matheus, 2024-01-03:
> @eder-matheus those are routing obstructions rather than placement blockages. I'm not sure this usage makes sense. Let's see how the discussion concludes before doing any work.

Yes, you're right. The code I've mentioned is not used, but we do consider macros to avoid using tracks that would be blocked by them.

closed 2025-08-12 by maliberty